ros2/rosidl #4: type support structure and dependencies

wjwwood, 2015-03-05:
Currently there is a single "type support" type for messages and a single one for services, `rosidl_message_type_support_t` and `rosidl_service_type_support_t` respectively. This structure is used to avoid introducing message type information into the majority of the `rmw` functions. For example there is one function which takes the message type and returns an instance of this type support structure, but the publish call does not need to have any information about the message type. It normally would, but instead it takes the type support structure and a `void *` pointing to the message. This is a very similar strategy to how DDS's API operates and allows us to a simpler `rmw` API with less code generation from macros, templates, or otherwise.

These types are currently defined in `rosidl_generator_c`, but used in both that package and the `rosidl_generator_cpp` package. This introduces a dependency on `rosidl_generator_c` for the `rosidl_generator_cpp` package. This bothers me a bit, but I'm not whether or not this actually matters. It seems cleaner to not have the C+ generator depend on the C one, but they might be so closely related that is doesn't actually matter.

I think that in the ideal situation, these type support types would be specified somewhere neutral and then the two generator packages could depend on the neutral package. One candidate would be in the `rmw` package, but then the namespace should probably change and it further couples `rmw` and the idl generators even more than they already are.

Another option would be to have a C++ version of these types and then both `rosidl_generator_c` and `rosidl_generator_cpp` will each define types for message and service type support. I think in this case it would make sense for them to be idiomatic and have different symbols so they are not used interchangeably. However, the problem with this approach is that you would then need C++ versions of any `rmw` function which takes the type support type as an argument.

Personally, I'm leaning towards doing nothing and leaving the dependency between the C and C++ generator packages, but I would also be ok with moving the type declaration to a neutral package.

Thoughts?

dirk-thomas, 2015-03-05:
I agree that currently there is no strong need to change it. But we can reconsider this once we have the full client library in C and C++ complete. Especially when the `rmi` will need to handle C as well as C++ ROS message instances.

wjwwood, 2015-03-05:
> Especially when the `rmw` API will need to handle C as well as C++ ROS message instances.

I thought that they would both be passed as `void *` and the contents of the message type support structure would inform the implementation about how to operate on them. Maybe that won't work though. We'll have to see. I'll close this for now.

commit 181227da (2015-11-25): fix introspection typesupport for messages without fields (fix #4)
closed 2015-03-05 by wjwwood