ros2/rosidl #8: On windows single element constructor lists fail to compile

tfoote, 2015-03-17:
The error is 

```
"C:\Jenkins\workspace\ros2_batch_ci_windows\build\userland\userland.sln" (default target) (1) ->
"C:\Jenkins\workspace\ros2_batch_ci_windows\build\userland\ALL_BUILD.vcxproj.metaproj" (default target) (2) ->
"C:\Jenkins\workspace\ros2_batch_ci_windows\build\userland\publisher.vcxproj.metaproj" (default target) (60) ->
"C:\Jenkins\workspace\ros2_batch_ci_windows\build\userland\publisher.vcxproj" (default target) (61) ->
(ClCompile target) -> 
  C:\Jenkins\workspace\ros2_batch_ci_windows\install\include\simple_msgs/AllDynamicArrayTypes_Struct.h(35): error C2668: 'std::vector<int8_t,std::allocator<_Other>>::vector': ambiguous call to overloaded function [C:\Jenkins\workspace\ros2_batch_ci_windows\build\userland\publisher.vcxproj]
  C:\Jenkins\workspace\ros2_batch_ci_windows\install\include\simple_msgs/AllDynamicArrayTypes_Struct.h(35): error C2668: 'std::vector<uint8_t,std::allocator<_Other>>::vector': ambiguous call to overloaded function [C:\Jenkins\workspace\ros2_batch_ci_windows\build\userland\publisher.vcxproj]
```

At issue is the constructor lists: 
```
  AllDynamicArrayTypes_()
  : my_bool({true, false, true})
  , my_byte({23, 42})
  , my_char({-2, 3})
  , my_float32({})
  , my_float64({-3.14l, 23.42l})
  , my_int8({-5})
  , my_uint8({5})
  , my_int16({1, 2, 3})
  , my_uint16({2, 3, 4})
  , my_int32({3, 4, 5})
  , my_uint32({4, 5, 6})
  , my_int64({5, 6, 7})
  , my_uint64({6, 7, 8})
  {
  }
```

The failing lines are int8 and uint8. Making them non-one length makes them pass. And making float32 one length causes the same failure.  

[Relevant Knowledgebase article for the error](https://msdn.microsoft.com/en-us/library/da60x087.aspx)

It works fine to pass the same initializer list as an argument in the constructor. ala
```
my_int8 = { -5 };
```

dirk-thomas, 2015-03-17:
Does the following satisfy the compiler on Windows?

    , my_int8({(int8)-5})

wjwwood, 2015-03-17:
I don't believe so, we actually tried that, but we can try it again more explicitly.

wjwwood, 2015-03-17:
Also, this error didn't come up for me when testing on my Windows machine, which is strange. There's more investigation to be done here, but we can likely implement a slightly ugly but function workaround using the in constructor assignment.

tfoote, 2015-03-17:
@esteve things that there might be a  workaround using {{x}}

wjwwood, 2015-03-18:
The double `{` solution seems to work on Windows. Thanks @esteve! I'll add that to the pull request and test it out on OS X / Linux.

wjwwood, 2015-03-18:
I suppose we need to actually make sure it does what we expect as well, I've only asserted that it compiles. These all compile: `{{}}`, `{{5}}`, `{{5, 6, 7}}`.

closed 2015-03-21 by wjwwood