ros2/rosidl #26: [rosidl_generator_c] fails on static array of non-primitive type

dirk-thomas, 2015-07-09:


ernestmc, 2015-10-20:
I would like to start working on this issue. How can I test the code to see this failure?

jacquelinekay, 2015-10-20:
@wjwwood will contact you about starting this issue

ernestmc, 2015-10-21:
I've added back the line referenced in #27 and when building everything I get the following error:
```
Generating C++ code for ROS interfaces
Traceback (most recent call last):
  File "/home/ecorbellini/repos/osrf/ros2_ws/install/lib/rosidl_generator_c/rosidl_generator_c", line 25, in <module>
    sys.exit(main())
  File "/home/ecorbellini/repos/osrf/ros2_ws/install/lib/rosidl_generator_c/rosidl_generator_c", line 20, in main
    args.generator_arguments_file,
  File "/home/ecorbellini/repos/osrf/ros2_ws/install/lib/python3.4/site-packages/rosidl_generator_c/__init__.py", line 53, in generate_c
    minimum_timestamp=latest_target_timestamp)
  File "/home/ecorbellini/repos/osrf/ros2_ws/install/lib/python3.4/site-packages/rosidl_cmake/__init__.py", line 89, in expand_template
    interpreter.file(h)
  File "/usr/lib/python3/dist-packages/em.py", line 2326, in file
    self.safe(scanner, done, locals)
  File "/usr/lib/python3/dist-packages/em.py", line 2368, in safe
    self.parse(scanner, locals)
  File "/usr/lib/python3/dist-packages/em.py", line 2388, in parse
    token.run(self, locals)
  File "/usr/lib/python3/dist-packages/em.py", line 1403, in run
    interpreter.execute(self.code, locals)
  File "/usr/lib/python3/dist-packages/em.py", line 2565, in execute
    exec(statements, self.globals)
  File "<string>", line 7, in <module>
KeyError: 'PrimitivesBounded'
Exception ignored in: <bound method Interpreter.__del__ of <empy pseudomodule/interpreter at 0x2b4086ab6b70>>
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/em.py", line 2063, in __del__
  File "/usr/lib/python3/dist-packages/em.py", line 2126, in shutdown
  File "/usr/lib/python3/dist-packages/em.py", line 2619, in finalize
  File "/usr/lib/python3/dist-packages/em.py", line 2169, in push
AttributeError: '_io.TextIOWrapper' object has no attribute 'push'
make[2]: *** [rosidl_generator_c/rosidl_generator_cpp/msg/bounded_array_bounded.h] Error 1
make[1]: *** [CMakeFiles/rosidl_generator_cpp__c.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 66%] Built target rosidl_generator_cpp__cpp
make: *** [all] Error 2
```
I believe that's the error that @dirk-thomas mentioned to occur in the template. I'll keep checking from here.

wjwwood, 2015-10-21:
@ernestmc That looks like a good place to start.

A more general improvement would be to have tests which generated a wide array of message types and tested them out, similar to: https://github.com/ros2/rosidl/blob/master/rosidl_generator_cpp/test/test_interfaces.cpp

ernestmc, 2015-11-04:
Traced error to
https://github.com/ros2/rosidl/blob/master/rosidl_generator_c/resource/msg__struct.h.template#L48
```
    print('ROSIDL_GENERATE_STATIC_ARRAY(%s, %s, %s, %s);' %
          (c_full_name, field.name, MSG_TYPE_TO_C[field.type.type], field.type.array_size))
```
for a non primitive type `field.type.type` is the name of the message that defines it and it fails as MSG_TYPE_TO_C does not contain that key.

dirk-thomas, 2015-11-04:
For a non-primitive types this line never gets executed because of the condition two lines above: https://github.com/ros2/rosidl/blob/master/rosidl_generator_c/resource/msg__struct.h.template#L46

ernestmc, 2015-11-05:
@dirk-thomas I understand that condition was added to avoid this issue. I took it off for my testings.
I'm not sure if the parser is the issue or the template should be modified for this case.

ernestmc, 2015-11-09:
Implemented the following fix and now it seems to be working:
https://github.com/ernestmc/rosidl/commit/cf4b21a77bc4e2585ecdc568ad284871de6476a4
I'll go on and implement some tests to check that the headers are generated correctly.

ernestmc, 2015-11-09:
After speaking with @wjwwood I left the [Type] parameter back in the macro and modified the template to handle separate cases for primitive and non primitive types. The appropriate type is passed in each case.
https://github.com/ernestmc/rosidl/commit/c6505491175729977ac6448e3ed61cae70d62b7d
https://github.com/ernestmc/rosidl/commit/13a11ea2779f25347d699727b7c094e523d36e77

wjwwood, 2015-11-10:
@ernestmc Those changes seem reasonable, can you push them to a branch on the rosidl repository and open a pull request? That way we can review the changes and run CI over them.

ernestmc, 2015-11-10:
Sure, I'll prepare the PR.
I'm working on a test for the generated C headers but I think we should open a new issue for that. Do you agree?

wjwwood, 2015-11-10:
Sure, either would work. If these changes fix a problem and the tests aren't ready, go ahead and open two pr's.

commit a37305ce (2015-11-12): Merge pull request #44 from ernestmc/issue_26

Fixes #26
closed 2015-11-12 by ernestmc