ros2/rosidl #46: Add tests to [rosidl_generator_c]

ernestmc, 2015-11-11:
Write some tests to check the types in the generated headers for different messages.
Similar to https://github.com/ros2/rosidl/blob/master/rosidl_generator_cpp/test/test_interfaces.cpp

tfoote, 2015-11-24:
We want to test, for each type, (primative, array, strings, nested, bounded/unbounded/fixed)
Test construction, loading, unloading, correctness, assertion, correct cleanup. @dirk-thomas please make into bulleted list

dirk-thomas, 2015-11-24:
The list of different types:
* bool
* byte
* char
* float32
* float64
* int8
* uint8
* int16
* uint16
* int32
* uint32
* int64
* uint64
* string (unbounded), string=N (fixed size), string<=N (upper bound)
* submessage
* each of them can be arrays (unbounded, fixed size, upper bound)

The things to test:
* get / set values, ensure valid value range works
* initialized correctly, default values
* alloc/free init/fini functions work
* memory handling is correct (e.g. manually running the test with varlgrind)

ernestmc, 2015-12-08:
@dirk-thomas while I was trying the tests on the string types I noticed that ```rosidl_generator_c__String__assignn``` doesn't check for the upper-bound limit on an assigned string. In fact it can increase the capacity over that size if the string passed is larger.

dirk-thomas, 2015-12-08:
Since `rosidl_generator_c__String__assignn` is a generic function it can't perform the check. In order to check for the upper boundary we would need to generate and use a message field specific function. Due to the increased complexity we didn't do that for now.

ernestmc, 2015-12-08:
Ok.

ernestmc, 2015-12-09:
@dirk-thomas here is what I commented yesterday. In the destroy function we free the pointer but we don't set it to NULL:
https://github.com/ros2/rosidl/blob/master/rosidl_generator_c/resource/msg__functions.c.template#L201-L208

dirk-thomas, 2015-12-09:
Because the pointer variable is local to this function there is no need to set it to zero. It goes out of scope on the next line anyway.

If we would like to set the pointer of the caller to zero we would need to pass a pointer to a pointer.

ernestmc, 2015-12-09:
Your right, the caller should manage it's pointer.

ernestmc, 2015-12-15:
Added tests for the following types:
* simple primitives
* simple primitives with default values
* strings
* unbounded arrays of primitives
* bounded arrays of primitives
* static arrays of primitives

The output using valgrind looks good:
```
$ valgrind --leak-check=yes ./test_interfaces_c
==27699== Memcheck, a memory error detector
==27699== Copyright (C) 2002-2013, and GNU GPL'd, by Julian Seward et al.
==27699== Using Valgrind-3.10.1 and LibVEX; rerun with -h for copyright info
==27699== Command: ./test_interfaces_c
==27699== 
Testing rosidl_generator_c message types...
Testing simple primitive message types...
Testing simple primitives with default values...
Testing string types...
Testing primitives unbounded arrays types...
Testing primitives bounded arrays types...
Testing primitives static arrays types...
All tests were good!
==27699== 
==27699== HEAP SUMMARY:
==27699==     in use at exit: 0 bytes in 0 blocks
==27699==   total heap usage: 35 allocs, 35 frees, 1,930 bytes allocated
==27699== 
==27699== All heap blocks were freed -- no leaks are possible
==27699== 
==27699== For counts of detected and suppressed errors, rerun with: -v
==27699== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
```

closed 2015-12-18 by wjwwood