ros2/rosidl #75: static_assert fails on test_interfaces.cpp

Karsten1987, 2016-01-31:
I am playing around with ROS2 and it seems that the static assertion check on [line 107](https://github.com/ros2/rosidl/blob/master/rosidl_generator_cpp/test/test_interfaces.cpp#L107) fails on my machine (x86, 14.04, 32bit)

```
primitive_size = 48
primitive_static_array_size = 140
```

any ideas on that?

wjwwood, 2016-01-31:
@Karsten1987 I guess it's likely a check that unintentionally assumes 64-bit (since we mostly use 64-bit machines for development), but I'm not certain without looking into the details.

@jacquelinekay do you have time to have a look at this? I believe it's some code you wrote, though I might be mistaken about that.

jacquelinekay, 2016-01-31:
I'll check it out. I wrote this a while ago and I recall trying to write it
in a portable manner, but I'm sure it could be improved.

jacquelinekay, 2016-01-31:
Huh, so, the test is originally written with the expectation that an array of structs of size 48 bytes with 3 entries will have size 48*3=144. However @Karsten1987 reports that his 32bit machine calculates the struct's size to be 140. My guess is that there is some optimization made for storing bools, which require less than a byte of representation, in this case. Additionally, if someone somewhere isn't being explicit about the sizes of data types, an int will be interpreted as 32 bits on 32 bit and 64 bits on 64 bits. BUT that wouldn't explain why the Primitives struct is the same size on both architectures and the array is 4 bytes smaller on 32 bit.

Unfortunately I'm not at my computer right now, so I can't experiment with the code.

jacquelinekay, 2016-02-01:
PrimitivesStatic must be at least 45 bytes (if bool is padded to 1 byte alignment).

sizeof(bool)+sizeof(byte)+sizeof(char)+sizeof(float32)+sizeof(float64)+sizeof(int8)+sizeof(uint8)+sizeof(int16)+sizeof(uint16)+sizeof(int32)+sizeof(uint32)+sizeof(int64)+sizeof(uint64) =
1+1+1+4+8+1+1+2+2+4+4+8+8=45

which on both architectures gets aligned to 48 bytes, because 48 is both a multiple of 4 and 8 (32 bits and 64 bytes).

Actually, on the 32 bit system the array of PrimitivesStatic objects is probably still 144 bytes (48*3), since the test on line 123 passes.

The struct where the static_assert fails is the message of only arrays of primitive types. The minimum size for this would be 3*45, which is 135. That gets aligned to the nearest multiple of 4 on the 32 bit system, which is 140, but it gets aligned to the nearest multiple of 8 on the 64 bit system, which is 144.

So the assumption of the test I wrote is wrong :) I'll write a patch.

jacquelinekay, 2016-02-02:
@Karsten1987 if you have time, can you test `master` of this package to check that it's fixed? Otherwise I'll set up a 32-bit VM.

closed 2016-02-02 by jacquelinekay

Karsten1987, 2016-02-02:
This does work, since I basically did the same modif locally at my machine in order to proceed with the compilation. thanks for the patch.

 jm2c: I believe it is noted somewhere on the tutorial page, but could be stressed out a bit more verbosely that 64-bit is assumed for the moment. I had similar problems with cam2image where most likely I linked against a wrong version of OpenCV. This is a different issue though.

jacquelinekay, 2016-02-02:
We could add a section to the wiki for Linux 32-bit. However, future support for 32-bit platforms might instead focus on cross-compiling for no OS or an embedded OS; it's something we need to give more thought and time to.

I'm surprised the package manager didn't install the i386 deb for OpenCV automatically. Do you have a source install of OpenCV? Did you resolve your problem and if so, how did you do it?

PR #76 (2016-02-02): Get rid of faulty assumptions in test
Fixes #75

As I recall, when I added this test I was trying to show that a fixed-size message struct is predictable at compile time. However, as issue #75 shows, the `sizeof` operator can give different outputs between compilers and architectures, because structure alignment is implementation defined, not defined by the language. http://stackoverflow.com/questions/119123/why-isnt-sizeof-for-a-struct-equal-to-the-sum-of-sizeof-of-each-member

If anyone has better ideas for improving this test, let me know. I was considering something fancy with `__attribute [... cut]