ros2/rosidl #77: review default constructors

tfoote, 2016-02-01:
There is a suggestion for unspecified default values to have the datatype's default value.

clalancette, 2017-04-25:
The default constructors not setting the default values for the member variables is starting to hurt.  For instance, see ros2/geometry2#18 , where valgrind ends up throwing a bunch of errors because the object members aren't initialized.  I've run into this at least twice now, and I'm guessing this is going to be a recurring stumbling block.  Thus, I think we should revisit this decision and set all of the values to have the default value.

dirk-thomas, 2017-04-25:
In ROS 1 we don't do this simply for performance. And for ROS 2 the rational is the same. There are cases where you don't want that initialization to cost performance since the common case is that you do set all fields of a message you want to use.

clalancette, 2017-04-25:
Do tests in ROS1 generally fill in all of the fields, then?  That's mostly where I've been running into the problem.

dirk-thomas, 2017-04-25:
Every code (test or not) should initialize all variables it is about to use.

clalancette, 2017-04-25:
Right, but the question is whether the initialization happens implicitly via a constructor, or explicitly.  It sounds like you are saying that the code in ROS 1 generally does it explicitly, since the constructors for messages do not default initialize all of the fields.  Is that the case?

dirk-thomas, 2017-04-25:
In ROS 1 we actually do initialize primtive fields (see last sentence in this section: http://wiki.ros.org/roscpp/Overview/Messages#Message_Generation).

clalancette, 2017-04-25:
OK, interesting.  So for ROS2 we have to make the trade-off between default initializing all fields (which is safer, more convenient, but slower), and sticking with the current status quo (which is more dangerous, but faster).  I'm not quite sure how we come to that decision.

Generally I'm on the side of making things safer for users, but I don't know how much of a performance cost default initialization will be.

clalancette, 2018-02-22:
This was actually done for ardent, so closing this out.

closed 2018-02-22 by clalancette