ros2/rosidl #99: build fails when only building FastRTPS

dirk-thomas, 2016-02-24:
http://ci.ros2.org/job/ci_windows/1092/

@esteve Can you please take a look at this since it seems to be related to the Python message generator.

jacquelinekay, 2016-02-24:
I believe this will be fixed by ros2/rosidl#95. Can you review?

dirk-thomas, 2016-02-24:
I triggered http://ci.ros2.org/job/ci_windows/1096/ using the branch of that PR.

jacquelinekay, 2016-02-25:
The CI job built. There are some Python test failures which look unrelated.

dirk-thomas, 2016-02-25:
With the referenced PR it at least passes the build step. But it still fails some Python related tests which shouldn't happen when only selecting FastRTPS.

jacquelinekay, 2016-02-25:
Good point. I haven't looked very carefully at the structure of rosidl_generator_py/rclpy but this line seems to be a problem:

https://github.com/ros2/rclpy/blob/master/rclpy/rclpy/__init__.py#L15

Shouldn't the middleware name be parameterized at least? Does rclpy only ever use Opensplice TS?

PR #95 (2016-03-09): Fail in rosidl_generator_py if no typesupport implementations found
Fixes #99
this is possibly a fix for python messages appearing to fail to build in ros2/common_interfaces#11
closed 2016-03-09 by jacquelinekay