ros2/rosidl #101: configure target language

firesurfer, 2016-03-01:
At the moment the cmake call rosidl_generate_interfaces(...) generated files for c and c++.
As far a I can see the c files aren't needed in case all projects using the generated files are written in c++.
How about adding an option to set the targeted languages? This would decrease the processing time.

wjwwood, 2016-03-01:
This is probably something you should configure system wide. It doesn't really make sense for a package to decide what languages it generates messages for, since it doesn't know what kinds of packages will utilize the messages it has defined.

I think it should be possible to just put an `AMENT_IGNORE` file in the folder for the generators you want to disable. Since they are not built, message packages which would normally generate code for each generator will not for the disabled generators. You'll have to clear out your workspace for this enable/disable to work, so it might make sense to have a configuration to disable generation (either a CMake variable or something else), but that should be determined by the generator in question and shouldn't be set or made an option by each message package explicitly.

firesurfer, 2016-03-01:
As far as I understood most packages are used in a clearly defined environment (e.g. a workspace) at development time.  So it wouldn't trouble if a package decides only to ship c++ files, because the developer knows what he did. When he wants to deploy his package he could still enable the build of other language files. 
From my point of view it would be nice to have set of standard generators I can use in a package but I don't have to (decrease build time or only targeting specific languages) (Especially regarding PR #100 )

wjwwood, 2016-03-01:
I agree, my point is that this should be configured workspace wide, not passed explicitly through a parameter to the `rosidl_generate_interfaces` cmake macro. Because once the limitation is set there, then someone needs to change your code to change the behavior. Rather I think it would be better for each generator to create a CMake option in each message package it operates which allows you to disable it. Then you can pass that CMake option to all the message packages (or all cmake packages even though that is overmatching quite a bit) which will prevent C messages, for example, from being generated. Another option would be to make these generators check an environment variable before continuing. Both options have their drawbacks, but I think they're still better than an option in the CMake code which whitelists the generators for just that package.

Ryanf55, 2023-06-14:
What about the cmake property [CMAKE_CXX_STANDARD_REQUIRED](https://cmake.org/cmake/help/latest/variable/CMAKE_CXX_STANDARD_REQUIRED.html#cmake-cxx-standard-required) and similar other language variables. If that is not set, then rosidl_generate_interface skips the C++ generation. 

Reference: https://crascit.com/2015/03/28/enabling-cxx11-in-cmake/

closed 2023-07-17 by firesurfer