ros2/rosidl #118: Consider narrowing the scope of `#undef ERROR` in generated message code

gerkey, 2016-04-18:
https://github.com/ros2/rosidl/commit/d4264b70387e91b8f46a40ff307881f28ff07c04#commitcomment-17146232

mikaelarguedas, 2017-08-07:
This is a known problem that we face is ROS1 as well (e.g. https://github.com/ros/geometry2/issues/172) and it would be great to see how we can address these collisions with Windows defined in the near future

dejanpan, 2017-10-27:
This issue has come up again here:

The generated file build\tf2_msgs\rosidl_generator_cpp\tf2_msgs\msg\tf2_error__struct.hpp contains the following code that conflicts with Windows' definition:
```
  // constants
  enum { NO_ERROR = 0u };
```

Windows SDK defines macro NO_ERROR in Error.h:
```
#define NO_ERROR            0
```

We have confirmed for myself that that culprit is Windows SDK. On a clean installation everything works OK but as soon as we installed Windows SDK the build fails in TF2 (now it fails in Release build too).

@wjwwood says:
`It might also be possible to change these constants in ROS 2 (with appropriate special handling in the ros1_bridge).`

tfoote, 2017-10-27:
This is also an issue with 'DELETE' defined in winnt.h: https://github.com/ros2/common_interfaces/issues/44

We'll either need some way to generically protect ourselves from some known set of conflicting strings or else have a blacklist of known strings that must be banned due to windows bad hygene in macro names. 

Might it make sense to have a known set of potentially colliding macros that automatically get undefed in our code generators if a conflict is detected?

dirk-thomas, 2018-02-27:
Options:
* undefine Windows specific MACRO
* name mangle the enum (e.g. appending an underscore)
* use different enum naming scheme (e.g. `kERROR`)

tfoote, 2018-04-30:
Related: https://github.com/ros2/design/pull/172