ros2/rosidl #124: Python generator called repeatedly in message interface packages

jacquelinekay, 2016-05-03:
The message "Generating Python code for ROS interfaces" is printed repeatedly during compilation of message interface packages. It seems the Python generator is being called repeatedly when it should only be called once.

dirk-thomas, 2016-05-03:
The code generation is also being retriggered during the installation step which rebuilds the code again.

mikaelarguedas, 2016-05-04:
#125 has fixed the multiple invocations of the python generator for Linux and OSX. It seems that on windows only the install step is fixed, the generator is still called multiple times during the build step.

dirk-thomas, 2016-05-04:
Do you have a link to a Jenkins job which shows that behavior?

mikaelarguedas, 2016-05-04:
This happens only on windows builds: http://ci.ros2.org/job/ci_windows/1268/
It may be linked to the fact the PythonExtra module cannot be found on windows because PEP-3149 doesn't specify anything for python extension suffixes on windows.
I haven't looked into it yet, I just noticed it on my CI jobs last night

dirk-thomas, 2016-05-04:
This referenced build is 8 days old so not really saying much. But a current build of master shows the same problem: http://ci.ros2.org/job/ci_windows/1295/consoleFull (search for "Generating Python", twice per package). There I will reopen this.

mikaelarguedas, 2016-05-04:
sorry messed up my copy paste. I was referring to the CI job of my last PR: http://ci.ros2.org/job/ci_windows/1294. But yeah same issue on current master.

mikaelarguedas, 2016-05-06:
Seems to be a cmake issue @dirk-thomas contacted kitware to find a solution/workaroud

dirk-thomas, 2016-05-12:
Still waiting for a feedback from Kitware / Microsoft on this.

PR #125 (2016-05-03): Fix generator being invoked multiple times
The reason the custom command was retriggered was that the files being promised to be generated included the Request / Response messages as well as the `__init__` files. But the generator didn't actually generate them.

Fixes #124.
PR #135 (2016-05-13): move custom command to a separate CMake file
After: http://ci.ros2.org/job/ci_windows/1333/consoleFull (search for "Generating Python" which only appears once for the `rosidl_generator_py` package).

Fixes #124.
closed 2016-05-13 by dirk-thomas