ros2/rosidl #170: [rosidl_parser] Parse strings default values in .msg files

mikaelarguedas, 2016-11-01:
Context: The [ROS 2 interface definition](http://design.ros2.org/articles/interface_definition.html) has more features than the ROS 1 one. Setting default values for your message fields directly in the `.msg` files is one of them. For now the parser doesn’t parse/escape default values for strings fields, so strings that contain [escape sequences](https://docs.python.org/2.0/ref/strings.html), [character substitution](https://docs.python.org/3.5/library/string.html#format-specification-mini-language) will cause bugs. e.g. 

```python
'foo\nbar'
'foo%sbar'
'foo'bar'
'foo{}bar'
```
So the goal is to escape every combination of characters using an additional `\` in front to prevent any breakage.

Task:
- [ ] Add an `escape_string` function [to rosidl_parser](https://docs.python.org/2.0/ref/strings.html)
- [ ] Call the `escape_string` function if the field type is a string [here](https://github.com/ros2/rosidl/blob/6c31eff290c6f7d15a78caef6b89074039e92a9d/rosidl_parser/rosidl_parser/__init__.py#L449)
- [ ] Extend the [parse value test](https://github.com/ros2/rosidl/blob/6c31eff290c6f7d15a78caef6b89074039e92a9d/rosidl_parser/test/test_parse_value_string.py) to test default values for all field types including strings with a set of known escape sequences

josephduchesne, 2017-02-12:
1. To clarify so I understand this clearly, the problem is that 'foo\nbar' will be read in as

> 'foo\nbar'

Then if character substitution is used, \n will be converted to a newline, when it should be left as the literals '\' and 'n', and escape_string should convert 'foo\nbar' to 'foo\\nbar' in this case, to prevent the undesired behavior?

2. Is the first step of the 2nd task removing " and type_.type != 'string'" from line 449? It's unclear to me what escaping the string in that context would do. 

> string[3] foo=["b\"ar", 'ba\'z', 'ba\nsh']

would get passed into parse_value_string as 

> ["b\"ar", 'ba,z', 'ba\nsh']

stripped to 

> "b\"ar", 'ba,z', 'ba\nsh'

split to

> "b\"ar"
> 'ba
> z'
> 'ba\nsh'

then spat out as

> b\"ar
> 'ba
> z'
> ba\nsh

I don't see where we would need escape_string, but I do see a need for some sort of alternative split function that's escape and quote aware to handle 'ba,z' being split incorrectly.

josephduchesne, 2017-02-12:
After further investigation, it seems that 'foo{}bar' should be escaped to 'foo{{}}bar', and 'foo%sbar' shouldn't need to be escaped, as long as everywhere uses the new .format() (effectively outlawing % interpolation, which is best practices in Python3).

This escaping raises a lot of questions. Shouldn't it be the responsibility of the code *using* potentially escaped strings  to call the escape function? If not, then it will become the responsibility of code using escaped strings to unescape them when the escaping is not needed, resulting in the need for an unescape_string method. Trying to maintain accuracy between escaping and unescaping 3 different sequences, any of which may be situationally unescaped by using .format or % seems like a path to madness.

Would a escape_string_for_format + escape_string_for_interpolation method make more sense?

Also, where is "foo'bar" breaking things? That seems like an issue with the thing that it is breaking, not the parser.

josephduchesne, 2017-02-13:
I have a branch with the following tests passing. Are these correct, or have I missed something?

```
    value = parse_primitive_value_string(
        Type('string'), "'foo'")
    assert value == 'foo'

    value = parse_primitive_value_string(
        Type('string'), '"\'foo\'"')
    assert value == "\\'foo\\'"

    value = parse_primitive_value_string(
        Type('string'), '"foo ')
    assert value == '\\"foo '

```

and

```
    value = parse_value_string(
        Type('string[]'), "[foo, 'bar', \"baz\"]");
    assert len(value) == 3
    assert value == ['foo', 'bar', 'baz']

    value = parse_value_string(
        Type('string[]'), "[foo, '\"bar\"', \"'baz'\"]");
    assert len(value) == 3
    print(value)
    assert value == ['foo', '\\"bar\\"', "\\'baz\\'"]
```

I'm only handling quote escaping so far, pending further clarification on where we want to go with the other escaping.

mikaelarguedas, 2017-02-13:
Hi @josephduchesne,
Thanks for your work and your interest in contributing to the ros2 project!

Sorry this issue has not been updated for a while. The parsing of string default value and escaping of `\` has been added to `rosidl_parser` in https://github.com/ros2/rosidl/pull/205.
Regarding the parsing of arrays of strings, there is a discussion and PR on the design repo to qualify how the parser should behave for string arrays: https://github.com/ros2/design/pull/111, and an outdated prototype that need to be revived to match the design doc https://github.com/ros2/rosidl/pull/197

Finally it has not be decided yet what escape sequences should be handled by the parser itself. I'll work on a proposal for the design doc to specify which escape sequences should be handled on the parser side, and how language specific generators should handle them after that.

I'll comment here and update this issue as soon as the design doc is ready.

josephduchesne, 2017-02-13:
Ok, excellent :) 

My code is very much 1 hour prototype, mostly so I could familiarise myself with ros2 development. I'm a big fan of the overall design changes from ros1 -> ros2 and would love to help accelerate its release. I'll look for some other outstanding issues to keep me busy in the mean time.

I suppose it's good that the cause of me finding the spec unclear was the result of incomplete spec, not my own lack of understanding. I'll toss my changes into a local branch and update it if they ever become relevant with the update design.

mikaelarguedas, 2018-06-14:
Fixed by https://github.com/ros2/rosidl/pull/197

closed 2018-06-14 by mikaelarguedas