ros2/rosidl #190: Deprecate char and byte and remap them as ROS 1 does

mikaelarguedas, 2016-12-13:


mikaelarguedas, 2016-12-13:
CI:
Linux: [image]
OSX: [image]
Windows: [image]

dirk-thomas, 2016-12-13:
For these changes the existing design docs need to be considered too. There are specific mappings for DDS as well as Python which don't have an equivalent if these would be removed. Therefore I think this needs further consideration and shouldn't be done for the beta 1.

mikaelarguedas, 2016-12-13:
I'm not advocating to get this in the beta. It would be great but maybe a bit premature.
I'm opening this so that we will have this and the corresponding design PRs opened for reference and will be able to move forward on it as soon as we decide that this is the way to go.

mikaelarguedas, 2017-01-05:
I never really replied to https://github.com/ros2/rosidl/issues/190#issuecomment-266643123:
- deprecating this means that we don't send the DDS type "char" anymore
- current implementation just remaps to int8/uint8 types. Meaning that in python the corresponding message fields have to be populated as numbers and not as `builtin.bytes` or `buitin.str`. Extra logic can be added in the python getter to allow both types and store the integral equivalent under the hood. I think the getters will always return the integral representation though.

Note: the fact that ROS1 consider `bytes` as **signed** goes against Python's definition of a `byte` that is an **unsigned octet**, that's why I stuck to only remap the types in this PR

mikaelarguedas, 2017-04-20:
CI:
Linux: [image]
OSX: [image]
Windows: [image]
aarch64: [image]

mikaelarguedas, 2017-04-26:
@mikaelarguedas update design accordingly

mikaelarguedas, 2017-05-18:
@mikaelarguedas need to update design to keep these types and offer flexibility for users to request a specific structure to receive them

johnmarkwayve, 2018-08-16:
It would be nice to have a type (perhaps `byte[]`) that maps to a python `bytes` object. Currently passing binary data around causes a large amount of unnecessary copying. E.g., if I want to use python to load an image from disk then send it to another `rclpy` node I have to: read it in as a `bytes` object, copy that data to a `list[int]` object, send the message (which presumably ends up packing the data back into an array of bytes), receive the message (which presumably ends up unpacking the data), then pack the data back into a `bytes` object.

It would also almost entirely clear up [ros2/ros2#509](https://github.com/ros2/ros2/issues/509)

mikaelarguedas, 2018-08-16:
Relevant discussion at https://discourse.ros.org/t/discussion-about-how-to-add-a-new-primitive-types-to-rosidl-msg/4607/3 as well.

dirk-thomas, 2019-03-12:
With the migration to use `.idl` files in ros2/rosidl#346 the types as defined in the IDL spec have been adopted. The upcoming article https://github.com/ros2/design/blob/gh-pages/articles/142_idl.md describes these types and the article https://github.com/ros2/design/blob/gh-pages/articles/143_legacy_interface_definition.md described the mapping from the `.msg` types.

closed 2019-03-12 by dirk-thomas