ros2/rosidl #208: 'Support for non-ASCII strings in python talker/listner messages

k-okada, 2017-03-14:
from https://discourse.ros.org/t/ros-2-alpha8/548/13

publishing multibyte message from cpp publisher is ok
```
I heard: [Hello World: 130]
I heard: [Hello World: 131]
I heard: [Hello World: 132]
I heard: [こんにちわ: 1]
I heard: [こんにちわ: 2]
I heard: [こんにちわ: 3]
```
But python publisher is not working
```
user@ubuntu-16:~/ros2_ws$ listener
I heard: [S0�0k0a0:]
I heard: [S0�0k0a0:]
I heard: [S0�0k0a0:]
^Csignal_handler(2)
user@ubuntu-16:~/ros2_ws$ listener_py 
python3: /home/user/ros2_ws/build/std_msgs/rosidl_generator_py/std_msgs/msg/_string_s.c:54: std_msgs_string__convert_to_py: Assertion `pydata != NULL' failed.
Aborted
```

mikaelarguedas, 2017-03-14:
Hi @k-okada, all Unicode objects in Python are converted into single byte strings when passed to the C layer of the ROS stack (RCL). Hence this is the expected behavior.
There is a proposal in progress to support multibyte strings in ROS2[1].
On the very short term I see 2 workarounds:
- use the c++ and std::string
- modify the Python message generator[2] to interpret strings as 2 or 4 bytes according to your needs

If you are willing to work on this we would be very happy to take your contribution as soon as the design document has been approved. Feel free to comment on the design PR itself, we would appreciate feedback from users using multibyte strings on a regular base.
[1]https://github.com/ros2/design/pull/117
[2]search for PyUnicode_1BYTE_DATA in https://github.com/ros2/rosidl/blob/master/rosidl_generator_py/resource/_msg_support.c.em

k-okada, 2017-03-16:
Thank you for your pointers, that really helps.
This is not urgent, just to contribute in return for give provide me information in discourse.
I have start reading the ros2/design#117 and also implementation https://github.com/ros2/rosidl/pull/209
I would be appreciate if you teach me where to put testcode for multibyte string communication.

mikaelarguedas, 2017-03-17:
Hi @k-okada,
Just tried out your PR, it looks great! And works back and forth mixing c++ and python :tada:.
Regarding testing, our communication tests are located in the [test_communication package](https://github.com/ros2/system_tests/tree/master/test_communication).
I think the best place to put multibyte test would be to add a new message fixture by adding a konnichiwa string in [the python fixtures](https://github.com/ros2/system_tests/blob/master/test_communication/test/message_fixtures.py#L97-L115).
something like:
```python
msg = Primitives()
msg.string_value = 'こんにちわ'
msgs.append(msg)
```
should be enough, no need to populate all the other fields of the message.
You will need to add the same message in the [c++ fixtures](https://github.com/ros2/system_tests/blob/master/test_communication/test/message_fixtures.hpp#L102-L123) for cross language communication to be tested as well.

I'll comment directly on the https://github.com/ros2/design/pull/117 for the design questions.

Thanks!

mikaelarguedas, 2017-08-07:
I'm going to close this in favor of the wider scope ticket https://github.com/ros2/ros2/issues/384

Thanks @k-okada for reporting and iterating on this

closed 2017-08-07 by mikaelarguedas