ros2/rosidl #213: Generation fails if message is in subsubfolder

firesurfer, 2017-04-11:
Hi,
I wanted to structure my messages in subfolders like this:
```
package
    |-CMakeLists.txt
    |-msg
            |-message1.msg
            |-subfolder1
                 |-message2.msg
```

My CMakeLists for generating the messages looks like this:
```
file(GLOB_RECURSE MESSAGE_FILES msg
    "*.msg"
)

rosidl_generate_interfaces(${PROJECT_NAME}
  ${MESSAGE_FILES}
  DEPENDENCIES builtin_interfaces ros2_components_msg
)
```

When calling ament build it fails with the error message:
`Interface file with unknown parent folder:`

dirk-thomas, 2017-04-11:
This is not a supported use case atm. The messages need to be in the subfolder `msg` and services in `srv`.

firesurfer, 2017-04-11:
What would it take to support this case? Is it planned to support it in the future?

dirk-thomas, 2017-04-11:
We are not planning to add support for the use case atm. The message generation code would need to be updated to allow that. Currently it relies on the folders to be `msg` / `srv`. The logic to decide which interface is a message and which is a message derived from service is very simple and doesn't require passing additional context information. Also it implicitly ensures that message names or service names are unique.

If PRs extend the CMake / Python interfaces involved in the message generation process we would certainly consider them to be integrated. But it should be made sure that this use case doesn't complicate each generator too much (trading features for code complexity).

firesurfer, 2017-04-11:
I will have a look at the message generation process. My idea is to treat messages in a subfolder of msg as messages. And services in a subfolder of srv as services. I don't think this would affect the uniqueness of the message names.

Only thing I'm wondering about is why the generation process doesn't use the fileextension to distinguish between messages and services.

mikaelarguedas, 2017-04-11:
> Only thing I'm wondering about is why the generation process doesn't use the fileextension to distinguish between messages and services.

It does use extensions to differenciate them. But each service generate a `Request` and a `Response` message that have an msg extension. That's where uniqueness of message name need to be ensured

dirk-thomas, 2017-04-11:
> I will have a look at the message generation process. My idea is to treat messages in a subfolder of msg as messages. And services in a subfolder of srv as services. I don't think this would affect the uniqueness of the message names.

A user can easily create two `.msg` files with the same name than which will conflict after message generation:

* `msg/foo/MyName.msg`
* `msg/bar/MyName.msg`

> Only thing I'm wondering about is why the generation process doesn't use the fileextension to distinguish between messages and services.

Because a service (e.g. `Foo.srv`) is expanded into two message definitions (`FooRequest.msg` and `FooResponse.msg`) but these `.msg` files still belong into the `srv` folder / namespace.

wjwwood, 2017-04-26:
We just discussed this in an issue triage meeting, and our team's consensus is that we will not spend time on this feature because of the issues @dirk-thomas pointed out above.

If you really want this feature, I'd suggest trying to justify why you need it and offer to open a pull request which addresses the concerns with respect to the potential collision of message files since the they don't share a folder.

firesurfer, 2017-04-26:
Perhaps you could give me a recommendation/alternative how to handle a package with a large number of message files? I thought the best way to do this is to put the message files into subfolders. Unfortunatly I got the requirement to put all messages into the same package.

wjwwood, 2017-04-26:
@firesurfer why can you not just put all the message files in the same folder?

I don't understand why that makes it easier to organize the code, not because I don't understand the hierarchy is useful for organizing, but because the folders would not affect the C++ header layout or the C++ namespaces. So you'd end up with message files in folders but you code (and others code) still have to use them from a flat namespace.

firesurfer, 2017-04-28:
Of course I can put all messages files in the same folder. It would work without any problems (That's what I'm actually doing at the moment). I just wanted to bring some structure into the messages on filesystem level, because over the time there will be more components added to the overall project and these components will need new message. So my thought was I could put them into subfolders. 

I actually do not care if they are put into extra namespaces or not. I will just have to easier to find the in the filesystem.

wjwwood, 2017-05-03:
> I just wanted to bring some structure into the messages on filesystem level, because over the time there will be more components added to the overall project and these components will need new message. So my thought was I could put them into subfolders.

I think that the position of the people at OSRF is that having a folder structure different from the namespace is not very useful unless it is matched in the generated code. Also, that the constraint of all message files for package needing to be in the same folder is useful, because it implicitly prevents messages with duplicate names.

However, if you see value in having additional hierarchy in the folder layout then we shouldn't stop you from doing it. So, I think we would take a pull request to allow that, but it needs to check for collisions manually, i.e. that you don't end up with something like this:

```
my_pkg
└── msg
    ├── arm
    │   └── Position.msg
    └── base
        └── Position.msg
```

Which if the constraint is relaxed would not be caught by any checks in our code currently.

nathangeorge1, 2018-02-26:
Hello,

I'd also like to note that I agree with @firesurfer that it'd be useful to be able to put the srv, msg, and action folders in another folder so that repo organization is better.

I understand the concern of conflicting names, but if this is a concern, I'd vote to have this be a compile-time error so that this feature can be realized, as I think it will not only help large codebases stay better organized but also novice coders understand that messages, services, and actions are all similar concepts.

I also understand that this is probably low priority in ROS2, as it doesn't seem crucial; but I'd like it :)

shubhamjain6594, 2025-02-02:
Commenting to check if this ever got into a distribution. 

As mentioned in https://github.com/UBCSailbot/custom_interfaces/pull/2#discussion_r1218933632, I am also trying to have the directory structure preserved as namespace. 

Issue - I want to move all my custom messages to a cenralized repository but I am loosing the context information by loosing the namespace info.

Any help/suggestion would be appreciated

firesurfer, 2025-02-07:
@shubhamjain6594 this was never fixed as far as I am aware of.

What you could do is have a centralized repository but just create sub-packages for each message "set" in it.
This is a rather common approach and makes it easier to separate messages into multiple repositories at a later point.