ros2/rosidl #216: Memory leak in nested messages

yamokosk, 2017-04-26:
This is really a continuation of issue ros2/rclpy#79. I am just continuing to go through top offenders reported by valgrind. I'm still using a simple pub/sub in Python and valgrinding the publisher. This is what I see:

```
==21797== 1,153 (1,152 direct, 1 indirect) bytes in 36 blocks are definitely lost in loss record 848 of 956
==21797==    at 0x4C2DB8F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==21797==    by 0xFBD4DC7: std_msgs__msg__MultiArrayDimension__create (multi_array_dimension__functions.c:43)
==21797==    by 0xF7B3457: std_msgs_multi_array_dimension__convert_from_py (_multi_array_dimension_s.c:15)
==21797==    by 0xF7B4423: std_msgs_multi_array_layout__convert_from_py (_multi_array_layout_s.c:50)
==21797==    by 0xF7ADF6F: std_msgs_float64_multi_array__convert_from_py (_float64_multi_array_s.c:37)
==21797==    by 0x9CF4B6A: rclpy_publish (_rclpy.c:282)
==21797==    by 0x4E9B9E: PyCFunction_Call (in /home/yamokosk/ros2_ws/ros2_env/bin/python3)
==21797==    by 0x524413: PyEval_EvalFrameEx (in /home/yamokosk/ros2_ws/ros2_env/bin/python3)
==21797==    by 0x528813: PyEval_EvalFrameEx (in /home/yamokosk/ros2_ws/ros2_env/bin/python3)
==21797==    by 0x528813: PyEval_EvalFrameEx (in /home/yamokosk/ros2_ws/ros2_env/bin/python3)
==21797==    by 0x52D82E: ??? (in /home/yamokosk/ros2_ws/ros2_env/bin/python3)
==21797==    by 0x529331: PyEval_EvalFrameEx (in /home/yamokosk/ros2_ws/ros2_env/bin/python3)

...

==21797== 1,408 (1,056 direct, 352 indirect) bytes in 33 blocks are definitely lost in loss record 866 of 956
==21797==    at 0x4C2DB8F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==21797==    by 0xFBD516F: std_msgs__msg__MultiArrayLayout__create (multi_array_layout__functions.c:41)
==21797==    by 0xF7B3DA3: std_msgs_multi_array_layout__convert_from_py (_multi_array_layout_s.c:18)
==21797==    by 0xF7ADF6F: std_msgs_float64_multi_array__convert_from_py (_float64_multi_array_s.c:37)
==21797==    by 0x9CF4B6A: rclpy_publish (_rclpy.c:282)
==21797==    by 0x4E9B9E: PyCFunction_Call (in /home/yamokosk/ros2_ws/ros2_env/bin/python3)
==21797==    by 0x524413: PyEval_EvalFrameEx (in /home/yamokosk/ros2_ws/ros2_env/bin/python3)
==21797==    by 0x528813: PyEval_EvalFrameEx (in /home/yamokosk/ros2_ws/ros2_env/bin/python3)
==21797==    by 0x528813: PyEval_EvalFrameEx (in /home/yamokosk/ros2_ws/ros2_env/bin/python3)
==21797==    by 0x52D82E: ??? (in /home/yamokosk/ros2_ws/ros2_env/bin/python3)
==21797==    by 0x529331: PyEval_EvalFrameEx (in /home/yamokosk/ros2_ws/ros2_env/bin/python3)
==21797==    by 0x52D2E2: ??? (in /home/yamokosk/ros2_ws/ros2_env/bin/python3)

...

==21797== LEAK SUMMARY:
==21797==    definitely lost: 2,287 bytes in 111 blocks
==21797==    indirectly lost: 393 bytes in 13 blocks
==21797==      possibly lost: 16,552 bytes in 26 blocks
==21797==    still reachable: 5,028,151 bytes in 3,504 blocks
==21797==         suppressed: 0 bytes in 0 blocks
==21797== Reachable blocks (those to which a pointer was found) are not shown.
==21797== To see them, rerun with: --leak-check=full --show-leak-kinds=all
```

My test code is using the leak fixes I have submitted in other PR's. So first on the destruction stack is the new generated destroy function I added:

```cpp
void std_msgs_float64_multi_array__destroy_ros_message(void * raw_ros_message)
{
  std_msgs__msg__Float64MultiArray * ros_message = (std_msgs__msg__Float64MultiArray *)raw_ros_message;
  (void)ros_message;

  std_msgs__msg__Float64MultiArray__destroy(ros_message);
}
```

Which then calls:

```cpp
void
std_msgs__msg__Float64MultiArray__destroy(std_msgs__msg__Float64MultiArray * msg)
{
  if (msg) {
    std_msgs__msg__Float64MultiArray__fini(msg);
  }
  free(msg);
}
```

Obviously the problem here is that `Float64MultiArray` is a nested message and its destroy function does not reflect this. I don't mind wading through one of the other code generators to fix this, but I wanted some feedback about where to put the fix. Does this go into `msg__functions.c.em`?

dirk-thomas, 2017-04-26:
Yes, the [destroy function](https://github.com/ros2/rosidl/blob/69400938eef1f6886fd65951d18fae7d3780e47d/rosidl_generator_c/resource/msg__functions.c.em#L201-L208) for a message needs to make sure that it also destroys all the non-primitive fields.

mikaelarguedas, 2017-04-26:
Thanks for reporting it and looking into it. IMO that was an oversight of https://github.com/ros2/rosidl/pull/215. As I see it the `_destroy_ros_message` function should have the same logic as the conversion functions, and for any nested message, free the message fields by calling their `_destroy_ros_message` function

yamokosk, 2017-04-26:
Absolutely right about the oversight in #215. I assumed that the *__destroy methods completely cleaned up.. even for nested messages. I guess I should have taken a cue from the complexity of the `convert_from_py` function.

I was debating were to locate this logic. So you think putting it in `_msg_support.c.em` is the right location?

dirk-thomas, 2017-04-26:
> I was debating were to locate this logic. So you think putting it in _msg_support.c.em is the right location?

Yes, absolutely.

mikaelarguedas, 2017-04-26:
> I was debating were to locate this logic. So you think putting it in _msg_support.c.em is the right location?

Yes I think that the right place to put it alongside the logic for the conversion. You should be able to reuse empy logic from convert_X_py to achieve it.

dirk-thomas, 2017-04-26:
If you have common code you can even move it into a function within the EmPy template.

yamokosk, 2017-04-27:
Looking for some assistance with this one. The problem is that I don't see how to do this without some sort of recursive function. The pattern I was going to copy was a for loop over the `spec.fields`:

```cpp
void @(spec.base_type.pkg_name)_@(module_name)__destroy_ros_message(void * raw_ros_message)
{
  @(msg_typename) * ros_message = (@(msg_typename) *)raw_ros_message;
  (void)ros_message;

@# Check each field in message. If it is a nested message, call its destructor
@[for field in spec.fields]@
@[  if not field.type.is_primitive_type() and not field.type.is_array]@
@{
nested_type = '%s__%s__%s' % (field.type.pkg_name, 'msg', field.type.type)
}@
  printf("destroying %s\n", "@(field.name)");
  @(nested_type) * @(field.name)_ros_message = &(ros_message->@(field.name));
  @(nested_type)__destroy(@(field.name)_ros_message);
@[  end if]@
@[end for]@

@# Now destroy the message we were told to destroy
  printf("destroying %s\n", "@(msg_typename)");
  @(msg_typename)__destroy(ros_message);
}
```

The above generates into:

```cpp
void std_msgs_float64_multi_array__destroy_ros_message(void * raw_ros_message)
{
  std_msgs__msg__Float64MultiArray * ros_message = (std_msgs__msg__Float64MultiArray *)raw_ros_message;
  (void)ros_message;

  std_msgs__msg__MultiArrayLayout * layout_ros_message = &(ros_message->layout);
  std_msgs__msg__MultiArrayLayout__destroy(layout_ros_message);

  std_msgs__msg__Float64MultiArray__destroy(ros_message);
}
```

However, it is obvious that this will only get the first layer of nested messages. Recursion would be great here, but I don't see how I can do that. Can the `spec` Python object be used to get the specs for other messages types?

dirk-thomas, 2017-04-27:
> However, it is obvious that this will only get the first layer of nested messages. Recursion would be great here, but I don't see how I can do that. Can the spec Python object be used to get the specs for other messages types?

By calling `@(nested_type)__destroy(..)` that function should already take care of its members. So it should already be recursive when you call the appropriate destroy function for each field.

yamokosk, 2017-04-27:
I might be missing something, but if that was the case, then simply calling 

```
void
std_msgs__msg__Float64MultiArray__destroy(std_msgs__msg__Float64MultiArray * msg)
{
  if (msg) {
    std_msgs__msg__Float64MultiArray__fini(msg);
  }
  free(msg);
}
``` 

should be enough to clear everything and my original PR shouldn't be leaking memory allocated in the messages. But as you can see, it calls free on the top level C message struct. And unless I misunderstand C's free method, its not smart enough to free any memory allocated to pointers in that struct.

This was my original question about where to locate this fix. Once place to do it would be by changing template code in `rosidl_generator_c` so that it does what I think you are saying. The other place is where I have been attempting the fix in `rosidl_generator_py`.

dirk-thomas, 2017-04-27:
The destroy function of a message `Foo` must clean up itself at the end but before it has to call the destroy function for each non-primitive member as you sketched in https://github.com/ros2/rosidl/issues/216#issuecomment-297847867

If that field of `Foo` happens to be another message it will take care of its recursive members. Once you modify the template to do that within the destroy function expanded for every message type the cleanup will be fully recursive.

allenh1, 2017-07-12:
closed, ros2/rosidl#220

closed 2017-07-12 by allenh1