ros2/rosidl #245: [rosidl_generator_c] Size of a ros_message (for memory allocation)

astralien3000, 2017-11-09:
Hello,

I would like to understand how we can get the size needed for memory allocation of a ros message from `rosidl_generator_c` generated code.
Indeed, when dealing with rclcpp, [the size (and type) of the message is transmitted by template](https://github.com/ros2/rclcpp/blob/master/rclcpp/include/rclcpp/node.hpp#L167).
But in C, [this information can only pass by the type_support](https://github.com/astralien3000/rclc/blob/master/rclc/include/rclc/rclc.h#L70).

I don't know how to allocate an empty message (for use by `rcl_take`) with only the type_support of rosidl_generator_c. [Here is my use-case.](https://github.com/astralien3000/rclc/blob/master/rclc/src/functions.c#L116-L118) I'm currently using an over-sized block of memory to do the job, but this is very bad.

Does someone have a solution for that ? 
If not, I guess it may be a feature request ?

Cheers

wjwwood, 2017-11-09:
The equivalent function in C++ for this is here:

https://github.com/ros2/rclcpp/blob/989084b3de84000a15fb4bd5a7d63929001cd769/rclcpp/include/rclcpp/subscription.hpp#L180-L187

Which in turn calls this:

https://github.com/ros2/rclcpp/blob/321e0b61b0988e906d3030aba073e1af18b40618/rclcpp/include/rclcpp/message_memory_strategy.hpp#L62

Which you've correctly deduced will allocate the data structure via the template argument. To do this in C you either need to pass the `sizeof` the type to the `rclc_create_subscription` call or we could extend the type support structure (https://github.com/ros2/rosidl/blob/34fdfe6b3b8b8de0c41afd079ae8f361747b3c4a/rosidl_generator_c/include/rosidl_generator_c/message_type_support_struct.h#L31-L36) to provide the sizeof the data type for you. I think the latter makes sense, as it would be useful potentially in other languages/cases.

mikaelarguedas, 2017-11-09:
> I think the latter makes sense, as it would be useful potentially in other languages/cases.

I agree, providing the sizeof in a typesupport structure became necessary in the past for typesupport introspection (see https://github.com/ros2/rosidl/pull/223/files) and I think providing it in the "general" message typesupport structure for reuse makes sense.

astralien3000, 2017-11-09:
Ah ? It is already in typesupport_introspection_c ?
Would it make sense to use it in a client library ?
I was understanding that rosidl_typesupport_* was intended to be used in rmw implementations while rosidl_generator_* was for client libraries. But [rclcpp seems to be linked to typesupport in some way](https://github.com/ros2/rclcpp/blob/master/rclcpp/CMakeLists.txt#L12-L13).

mikaelarguedas, 2017-11-09:
Sorry this was not clear in my comment:
I agree that we should add it to the `rosidl_message_type_support_t` structure (that is used by the client libraries regardless of which typesupport is used).
The `typesupport_introspection` is a specific case use by _some_ rmw_implementations, I just wanted to point out that there is a precedent (of providing the size through a type support structure) and that it makes sense to me to also expose this to the client library through the `rosidl_message_type_support_t` structure rather than by calling `sizeof()` on it before passing it to various functions

astralien3000, 2017-11-09:
Ok, so yes adding the size to `rosidl_message_type_support_t` seems to be the best option. I will try to do a pull request tomorrow about that, if no one have done that beforehand (tell me so 😉).

astralien3000, 2017-11-09:
Well, I just got an idea.
Instead of adding this to rosidl_generator_c, I could "simply" pass an other structure than `rosidl_message_type_support_t`...

Indeed, [instead of doing that](https://github.com/astralien3000/examples/blob/master/rclc/minimal_subscriber/not_composable.c#L35), I could do : 
```c
struct rclc_type_support_t {
    rosidl_message_type_support_t * type_support;
    size_t size_of;
};
// ...
#define RCLC_GET_MSG_TYPE_SUPPORT(pkg, dir, msg) \
    ((rclc_type_support_t){ \
        ROSIDL_GET_MSG_TYPE_SUPPORT(pkg, dir, msg), \
        sizeof(pkg##__##dir##__##msg) \
    })
// ...
rclc_subscription_t* sub =
    rclc_create_subscription(node, RCLC_GET_MSG_TYPE_SUPPORT(std_msgs, msg, String), "topic", ...);
```

wjwwood, 2017-11-10:
Yeah, using an rclc specific type support structure which also has a size of field populated by an rclc specific type support macro is a good option too.

astralien3000, 2017-11-10:
I think I will go for that instead of modifying rosidl, at least for the moment. If you want to debate this point, we can continue this issue, I will follow the choice that would be made. But for my initial problem, the issue can be closed.

Thank you for your help !

closed 2017-12-07 by astralien3000