ros2/rosidl #259: [rosidl_generator_cpp] add <<operator for message classes

mikaelarguedas, 2018-01-08:
We should provide a `<<operator` like in ROS 1 to ease printing to streams. Some places in the codebase print manually message content to stream in the meantime, e.g. https://github.com/ros2/demos/blob/144fb6d80810f149fa2457ae8542b332a6497674/demo_nodes_cpp/src/parameters/parameter_events.cpp#L26

mikaelarguedas, 2018-02-06:
this will require a decision on what format (YAML, JSON, something else?) we want for the output. 
Other PR related to the formats chosen: https://github.com/ros2/rclcpp/pull/275

dirk-thomas, 2018-02-06:
The decision from the last conversation on this topic was to *not* create such operators. Instead there should be global functions specialized on the message named `to_yaml` / `to_json` / `to_whatever_you_like`.

wjwwood, 2018-02-06:
I didn't think that we ruled out an `operator<<` which defaulted to one of those, for convenience.

PR #523 (2020-09-10): add to_yaml() function for C++ messages
Closes #259. Instead of defining a `<<` operator on the message this patch adds a `to_yaml()` function taking a message as well as an `ostream`. That allows to implement other formats like `to_xml()` side by side.

Atm all logic is in header files since `rosidl_generator_cpp` doesn't create a library.

CI builds testing `rosidl_generator_cpp`:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]
closed 2020-09-10 by dirk-thomas