ros2/rosidl #265: Usage examples for introspection

janismac, 2018-02-28:
Are there any usage examples for `rosidl_typesupport_introspection_cpp`? I managed to call `rosidl_typesupport_introspection_cpp::get_message_type_support_handle<my_package::msg::MyMessage>();`. But I'm not sure how to extract the IDL information. I could reverse-engineer my way through the generated `*__type_support.cpp` file. But that might lead to some brittle/unsafe code.

dirk-thomas, 2018-02-28:
You might want to look at the structs defined by the C++ introspection typesupport: https://github.com/ros2/rosidl/blob/15609e2ee204cfbb2b024de3552b9c806e152111/rosidl_typesupport_introspection_cpp/include/rosidl_typesupport_introspection_cpp/message_introspection.hpp

These should give you some idea what information is being made available.

The best example using the C++ introspection typesupport can be found in the RMW implementation of FastRTPS which uses this to programmatically convert the message types: https://github.com/ros2/rmw_fastrtps/blob/9438cca2a6a21b2436684607fea8a78624363f80/rmw_fastrtps_cpp/include/rmw_fastrtps_cpp/TypeSupport_impl.hpp

janismac, 2018-03-01:
Thanks! My biggest problem was recovering the `std::vector<T>` from just a `void*`, since `T` is unknown. The `reinterpret_cast<std::vector<unsigned char> *>(field)` is a good compromise, I think I'll use that. 

One more question: Why is `rosidl_message_type_support_t.data` of type `void*` and not of type `MessageMembers*`?

dirk-thomas, 2018-03-02:
The struct `rosidl_message_type_support_t` is defined in the `rosidl_generator_c` package. The `data` field can contain arbitrary information.The `MessageMembers` are e.g. defined in the introspection typesupport and are specific to that package. So the data field can't have that type since:

* `rosidl_generator_c` does not depend or know about the introspection typesupport package
* different type support implementations store different information in the data field and use their own custom type

closed 2018-03-03 by janismac