ros2/rosidl #272: Using C++ keywords as fieldnames results into compilation error

firesurfer, 2018-04-25:
I realised today that using C++ keywords as fieldnames results into a compilation error.
Example:

```
bool delete false
```
Will result into:
```
rosidl_generator_cpp/iboss_msgs/msg/request_stored_data__struct.hpp:90:16: error: expected member name or ';' after declaration specifiers
  _delete_type delete;
```

It would be great if either the message generator would throw an error like: Don't use c++ keywords or 
the message generator would somehow escape these fields.
If this is not a viable solution for you I think a hint in the documentation would be useful:
https://github.com/ros2/ros2/wiki/About-ROS-Interfaces

dirk-thomas, 2018-04-25:
> the message generator would somehow escape these fields.

Absolutely, that is what should happen in each language message generator specific to the keywords in that language.

Related to #118.

firesurfer, 2018-04-25:
Just for information: I did a short search and apparently it isn't possible in C++ to escape keywords. The usage of keywords as identifiers is forbidden.

dirk-thomas, 2018-04-25:
> it isn't possible in C++ to escape keywords

I am not sure I understand your comment. If the `delete` field would result in a C++ member `delete_` it doesn't collide with the keyword anymore. This step is commonly called "mangling" but I assumed what is what you referred to as "escaping".

firesurfer, 2018-04-25:
I thought of escaping in the way of doing someting like ` int @class = 0` what would be possible in C#. 
Changing a field named `delete` to  `delete_` it would work of course in C++.

tfoote, 2018-04-30:
I've started an design article on this topic at: https://github.com/ros2/design/pull/172

firesurfer, 2023-07-17:
I guess this can be closed @dirk-thomas

closed 2023-07-17 by firesurfer

Rockjack00, 2025-04-15:
I see that this issue was closed a few years ago, but I am running into the same problem.  I don't see a change to the ROS 2 docs about [field names](https://docs.ros.org/en/rolling/Concepts/Basic/About-Interfaces.html#field-names) in interfaces.

> I've started an design article on this topic at: [ros2/design#172](https://github.com/ros2/design/pull/172)
It seems like mangling the fields was discussed here but never implemented.

Was this issue ignored or should it be reopened?