ros2/rosidl #280: is there any complete name rules for name of *.msg file and variable in *.msg file when building msg idl ?

lxbeyond, 2018-05-29:
when I build my msg file, I found some name rules of creating msg file, such as : 
rule 1: there should't be any upper-case characters in varialbe name in msg file. for example, 
      "string abc" is valid,  and "string aBc" in invalid.
rule 2: the name of msg file should begin with upper-case characters. for example, msg file name     
      "Tmpmsg.msg" is valid, but "tmpmsg.msg" is invalid.
so I wonder whether if any complete rules for msg naming which will make me do less trying work when I want to build a correct msg ?

lxbeyond, 2018-05-29:
in addtion , rule 3 for variable defining in msg file is that:
there should be only one space between type and variable name. for example,"int64 col" is valid,but "int64       col" is invalid

mikaelarguedas, 2018-05-29:
The design document defining the ROS interfaces is available here: http://design.ros2.org/articles/interface_definition.html

The naming rules as well as msg/srv file formatting are detailed in the "Conventions" section.

The rule 3 you declared above is not part of the conventions: `While a single space is mandatory to separate tokens additional spaces can be inserted optionally between tokens.`.
I just tried on existing msg files and couldn't get them to fail when using multiple spaces between the field type and the field name. Are you sure you added only spaces between them ? Can you provide a msg file that fails for you so that we can reproduce the issue ?

lxbeyond, 2018-05-30:
I did't make it clear about rule 3, there should not be table between type and variable. "int64 \t col" is invalid

lxbeyond, 2018-05-30:
and i find a special msg [](https://github.com/ros2/common_interfaces/blob/master/sensor_msgs/msg/PointField.msg)
This message holds the description of one point entry in the
# PointCloud2 message format.
uint8 INT8    = 1
uint8 UINT8   = 2
uint8 INT16   = 3
uint8 UINT16  = 4
uint8 INT32   = 5
uint8 UINT32  = 6
uint8 FLOAT32 = 7
uint8 FLOAT64 = 8

string name      # Name of field
uint32 offset    # Offset from start of point struct
uint8  datatype  # Datatype enumeration, see above
uint32 count # How many elements in the field

what dose the variable name INT32 mean ? is it the same as variable "name" and "offset" ? if so ,it  violates the rule 1 .

mikaelarguedas, 2018-05-30:
> I did't make it clear about rule 3, there should not be table between type and variable. "int64 \t col" is invalid

Yes, as the specification states, only whitespaces is allowed to separate tokens

> what dose the variable name INT32 mean ?

This is a constant. From the design document linked above:
"Constant names must be uppercase alphanumeric characters with underscores for separating words. [...]"

>  is it the same as variable "name" and "offset" ?

The name of that constant is `INT32`, the other variables you refer to are named `name` and `offset`, so they have different name.

I believe the original question has been answered. Can this issue be closed ?

If you have further questions, [we encourage you](https://discourse.ros.org/t/please-ask-all-questions-on-ros-answers/4259) to ask them on [answers.ros.org](https://answers.ros.org/questions/) as the entire ROS community will see them and respond in a timely manner.
Thanks!

lxbeyond, 2018-05-31:
I meant can we refer to the data "INT32" in the way "varia.INT32"  the same as "varia.offset" ?

mikaelarguedas, 2018-05-31:
This depends on the language you use.

You can see the design documents defining how msg files are translated into datastructure for each language:
- [Generated C++ interfaces](http://design.ros2.org/articles/generated_interfaces_cpp.html)
- [Generated Python interfaces](http://design.ros2.org/articles/generated_interfaces_python.html)

The C++ one says: "Numeric constants are defined as enums within the struct." So they need to be accessed as enums
The Python one says: "A constant is defined as a class variable with uppercase name." so it can be accessed as a variable.

mikaelarguedas, 2018-05-31:
Closing this as I believe the original question has been answered.

@lxbeyond please ask any follow-up question on  [answers.ros.org](https://answers.ros.org/questions/) so that the answers are easily discoverable by other users

closed 2018-05-31 by mikaelarguedas

lxbeyond, 2018-07-04:
Recently we found a strange thing when building msg file:
in case one, we build one msg file whose name is "MyTest.msg", whatever content in it , which just build well,
in case two ,  we want to build another msg file whose name is "MYTEST.msg", whatever content in it , which still build well.
but as ”http://design.ros2.org/articles/interface_definition.html“ said "names must use an upper camel case", why case 2 work?
test rep: https://github.com/lxbeyond/ros2msgtest

mikaelarguedas, 2018-07-04:
@lxbeyond Please ask questions on [answers.ros.org](https://answers.ros.org)