ros2/rosidl #297: Redefinition of typedef "rosidl_xxx_type_support_t

BorjaOuterelo, 2018-09-18:
Hi

I am getting some warnings compiling on Linux.

- Operating System:
  - ArchLinux 4.18.7
- Git rev:
  - 5cdd36e02e0ad66a7652feed03bede33466413cf (HEAD, tag: 0.5.1, origin/bouncy)
- Compiler:
  - gcc version 8.2.1

Compiling this package using -Wpedantinc on GCC outputs some typedef redefinition issues:

```console
/install/rosidl_generator_c/include/rosidl_generator_c/service_type_support_struct.h:37:3: warning: redefinition of typedef ‘rosidl_service_type_suppor
t_t’ [-Wpedantic]                                                                                                                                                                             
 } rosidl_service_type_support_t;                                                                                                                                                             
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                                                                                                                              
/install/rosidl_generator_c/include/rosidl_generator_c/service_type_support_struct.h:27:46: note: previous declaration of ‘rosidl_service_type_support_
t’ was here                                                                                                                                                                                   
 typedef struct rosidl_service_type_support_t rosidl_service_type_support_t;
```

and

```console
/install/rosidl_generator_c/include/rosidl_generator_c/message_type_support_struct.h:36:3: warning: redefinition of typedef ‘rosidl_message_type_suppor
t_t’ [-Wpedantic]                                                                                                                                                                             
 } rosidl_message_type_support_t;                                                                                                                                                             
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                                                                                                                              
/install/rosidl_generator_c/include/rosidl_generator_c/message_type_support_struct.h:26:46: note: previous declaration of ‘rosidl_message_type_support_
t’ was here                                                                                                                                                                                  
 typedef struct rosidl_message_type_support_t rosidl_message_type_support_t;
```

Shouldn't be those structs just a struct definition without typedef them?.

```C
typedef struct rosidl_message_type_support_t rosidl_message_type_support_t;

typedef const rosidl_message_type_support_t * (* rosidl_message_typesupport_handle_function)(
  const rosidl_message_type_support_t *, const char *);

struct rosidl_message_type_support_t
{
  const char * typesupport_identifier;
  const void * data;
  rosidl_message_typesupport_handle_function func;
};
```
instead of 

```C
typedef struct rosidl_message_type_support_t rosidl_message_type_support_t;

typedef const rosidl_message_type_support_t * (* rosidl_message_typesupport_handle_function)(
  const rosidl_message_type_support_t *, const char *);

typedef struct rosidl_message_type_support_t
{
  const char * typesupport_identifier;
  const void * data;
  rosidl_message_typesupport_handle_function func;
} rosidl_message_type_support_t;
```

I can create a pull request for this if you are OK.

Best.

wjwwood, 2018-09-18:
Yeah, that sounds right. Pull request would be appreciated.

dirk-thomas, 2018-10-17:
Addressed by #299.

closed 2018-10-17 by dirk-thomas