ros2/rosidl #303: Generate C++ typesupport for actions

sloretz, 2018-10-18:
## Feature request

There should be a c++ file importable by `#include <package_name/action/actionname.hpp>` which contains everything needed to use an action type in c++.

#### Feature description

It needs to be possible to generate a header file that describes an action, and has all the information needed to create topics and services. Since services already are an example of a typesupport that needs to include messages, actions should use the same strategy.

#### Implementation considerations

##### Output files

Services have a top level header which includes two files: `traits` and `struct`. Traits indicates whether either the request or response is bounded or unbounded in size. I don't think there needs to be a traits file for the whole action.
https://github.com/ros2/rosidl/blob/206dd2bbea68ece80d5e811236cb4aecda495d6c/rosidl_generator_cpp/resource/srv.hpp.em#L22-L23

The struct file includes the request and response message, as well as defines a struct with a `using` directive that indicates which message is which

https://github.com/ros2/rosidl/blob/206dd2bbea68ece80d5e811236cb4aecda495d6c/rosidl_generator_cpp/resource/srv__struct.hpp.em#L31-L35

The action struct template could look something like this

```c++

#include "@(spec.pkg_name)/action/@(get_header_filename_from_msg_name(spec.action_name))__sendgoal.hpp"
#include "@(spec.pkg_name)/action/@(get_header_filename_from_msg_name(spec.action_name))__get_result.hpp"
#include "@(spec.pkg_name)/action/@(get_header_filename_from_msg_name(spec.action_name))__feedback.hpp"
// Convenience include for common service and message definition
#include <rcl_interfaces/srv/cancel_action.hpp>
#include <rcl_interfaces/srv/goal_status.hpp>

 struct @(spec.action_name) 
 { 
   using SendGoalService = @(spec.pkg_name)::action::@(spec.action_name)_SendGoal; 
   using GetResultService = @(spec.pkg_name)::action::@(spec.action_name)_GetResult; 
   using FeedbacMessage = @(spec.pkg_name)::action::@(spec.action_name)_Feedback; 
};
```

##### CMake targets

The CMake target needs to be aware of the generated output files so that it can rebuild stuff downstream when they change. Awareness of the files generated by `.action` needs to be included here

https://github.com/ros2/rosidl/blob/206dd2bbea68ece80d5e811236cb4aecda495d6c/rosidl_generator_cpp/cmake/rosidl_generator_cpp_generate_interfaces.cmake#L25-L36

connects to ros2/ros2#583

apojomovsky, 2018-11-01:
Hi @sloretz , I have a little doubt here, just want to ensure I got it right: From the explanation above, I understand that we don't need to generate trait files for actions (i.e. expand from an `action__traits.hpp.em`). Is that correct?

sloretz, 2018-11-13:
Fixed by #312 and ros2/rosidl_typesupport#36

closed 2018-11-13 by sloretz

serroukha, 2022-03-30:
Hello!! how can i know the right name of the header file name ??