ros2/rosidl #305: Parse .action files

sloretz, 2018-10-18:
## Feature request
It needs to be possible to parse `.action` files into an intermediate representation.

#### Feature description
`rosidl_parser` needs methods `parse_action_file` and `parse_action_string` which read a `.action` file and generate an intermediate representation that can be used by downstream generators.

The intermediate representation should have service definitions for the `Send Goal` and `Get Result` services, as well as a message definition for the `Feedback` topic. 

#### Implementation considerations

`parse_service_string` Splits the service file by `---` and calls `parse_message_string` on the individual parts. The code for actions should create separate strings for the services and message and then call `parse_service_string` and `parse_message_string` to create a description of an action.

https://github.com/ros2/rosidl/blob/206dd2bbea68ece80d5e811236cb4aecda495d6c/rosidl_parser/rosidl_parser/__init__.py#L686-L705

connects to ros2/rose#583

sservulo, 2018-10-24:
@sloretz what's the route we're taking on action file structure regarding service definition, a more explicitly defined action file? E.g. for the fibonacci action:
```
# Goal service
int32 order
string/int32 uuid
---
bool accepted
time timestamp
===
# Result service
string/int32 uuid
---
string status
int32[] sequence
===
# Feedback topic
int32[] sequence
```
or a more implicit one:
```
# Goal
int32 order
===
# Result
int32[] sequence
===
# Feedback
int32[] sequence
```

sloretz, 2018-10-24:
@sservulo Same as ROS 1 http://wiki.ros.org/actionlib#A.action_File to make it easier to port them

```
# Define the goal
uint32 dishwasher_id  # Specify which dishwasher we want to use
---
# Define the result
uint32 total_dishes_cleaned
---
# Define a feedback message
float32 percent_complete
```

sservulo, 2018-10-24:
@sloretz Got it, thank you.

PR #311 (2018-11-07): Action file support
This PR closes #305. Added `.action` file parsing to rosidl_parser.
CMake preprocessing of .action files into  services and messages is coming to this branch from `sloretz/speficication-to-str`.
closed 2018-11-07 by sservulo