ros2/rosidl #316: heuristic to extract comments from .msg files

dirk-thomas, 2018-11-09:
Related to #298.

While the file format of `.msg` files contains comments it isn't expressive enough to correlated them to specific parts of the file. In order to maintain the information from the comments when transforming the interface definition to `.idl` files (either transparently during the build or explicitly by the user) a heuristic can be applied to at least guest the desired correlation and include the comments in the resulting `.idl` file.

While that heuristic might not work in all existing cases perfectly it should allow that existing `.msg` could be updated to match the heuristic. The heuristic should also be easy to document as well as implement.

The following distinguishes three kind of comments:

* a "full comment line" line starting with a `#`
* an "indented comment line" starting with whitespace followed by a `#`
* a "trailing comment" starting with an actual definition (a constant or field) followed by a `#`

The proposed heuristic:

1. The first N consecutive "full comment lines" are correlated to the message itself. E.g.:

   ```
   # Message
   # specific
   # comment
   
   # unrelated
   ```

   ```
   # Message
   # specific
   # comment
       # unrelated
   ```

   ```
   # Message
   # specific
   # comment
   
   bool foo  # unrelated
   ```

2. All full comment lines not part of the message level are used for the first definition.

3. A "trailing comment" is associated to the definition made on the same line.

4, An "indented comment line" is associated to the *previous* definition (if there has been one). Otherwise it is being ignored.

These comments are then being transformed into an annotation in the `.idl` file.

---

The comment lines correlated to a definition can additionally be scanned for known patterns. For now only a single one is proposed here:

1. If the comment lines of a definition contain a string like `[...]` the content between the brackets is considered to be a unit. To avoid the misinterpretation of ranges the inner string must not contain a comma or dash. The detected range is then transformed into a `@unit` annotation in the `.idl` file and removed from the comment.

I am looking for early feedback in the proposal.

clalancette, 2018-11-13:
I like the general idea here.  One question that is not explicitly covered above is about "full comment lines" *after* the first field.  If we have:

```
# a field comment
bool a_field
# b field comment
bool b_field
```

Will "a field comment" get associated to `a_field`, and "b field comment" associated to `b_field`?  Or would the latter be ignored?  Also, how about commented-out fields like:

```
bool a_field
#bool b_field
bool c_field
```

How would that be handled?

dirk-thomas, 2018-11-13:
> Will "a field comment" get associated to `a_field`, and "b field comment" associated to b_field?

Yes.

> how about commented-out fields like:

Field `c_field` will have a comment "bool b_field" since there is no way to distinguish the content of the comment.

clalancette, 2018-11-13:
> Field `c_field` will have a comment "bool b_field" since there is no way to distinguish the content of the comment.

Great, that's what I thought.  It wasn't clear from the initial description what would happen to commented lines after the first field.  In that case, I'm fine with the proposal here; we'll have to fix up some of our `.msg` files (depending on whether we keep them as `.msg` or hand-convert them to `.idl`), but that should be fine.

> 1. **If the comment lines of a definition contain a string like `[...]` the content between the brackets is considered to be a unit.** To avoid the misinterpretation of ranges the inner string must [...]

What is the `@unit` annotation in IDL used for?  I'm somewhat concerned that this heuristic is going to trigger something bad on downstream packages, but if it is just documentation, then I'm fine with that.

dirk-thomas, 2018-11-13:
> What is the `@unit` annotation in IDL used for?

It only the unit of the field value, e.g. `rad` or `s` or `lbs`. It doesn't have any functional effect atm.

sloretz, 2018-11-13:
> The first N consecutive "full comment lines" are correlated to the message itself.

In [this message](https://github.com/ros2/common_interfaces/blob/master/diagnostic_msgs/msg/KeyValue.msg) does the heuristic consider the first line to be message level or field level?

```
# What to label this value when viewing.
string key
# A value to track over time.
string value
```

sloretz, 2018-11-13:
>> does the heuristic consider the first line to be message level or field level?
>
> Message level.

Three are a few messages where the first comments are for the first field.
https://github.com/ros2/common_interfaces/blob/master/actionlib_msgs/msg/GoalID.msg
https://github.com/ros2/common_interfaces/blob/master/visualization_msgs/msg/InteractiveMarker.msg

I like that the heuristics so far are simple and only require one pass on the message. Maybe adding another heuristic that if all the lines in-front of the first field are consecutive comments and

sloretz, 2018-11-13:
How important is avoiding whitespace changes to existing messages? If I understand correctly messages with field comments but no message comment ([ex1](https://github.com/ros2/common_interfaces/blob/master/actionlib_msgs/msg/GoalID.msg), [ex2](https://github.com/ros2/common_interfaces/blob/master/visualization_msgs/msg/InteractiveMarker.msg)) will need to start with a blank line once this heuristic is implemented.

dirk-thomas, 2018-11-13:
> How important is avoiding whitespace changes to existing messages?

I would say the heuristic tries to do the right thing for a lot of existing messages as well as be easy to understand.

> If I understand correctly messages with field comments but no message comment (ex1, ex2) will need to start with a blank line once this heuristic is implemented.

Correct (if you care about the correct extraction of the comments). I think this is the case for either options (either consider the line to be part of the message or the first field). There are existing messages for both cases - some where the line describes the message - some where the line describes the first field.

dirk-thomas, 2018-11-14:
See ros2/common_interfaces#63 for an example on how to update existing message definitions to match the heuristic "better".

jacobperron, 2018-11-14:
> An "indented comment line" is associated to the previous definition

Considering the following case:

```
# a_field
bool a_field
# b_field
    # more about b_field
bool b_field
```
My understanding is "more about b_field" will be associated with `a_field`.
What about preferring association of indented comments to both the previous defintion **or** previous comment, whichever comes first?

Perhaps this example is too unusual and we shouldn't bother trying to handle it. In any case, it might be good to document the expected behavior.

dirk-thomas, 2018-11-14:
> Considering the following case:

Can you point to an existing message file which uses this style?

dirk-thomas, 2018-11-15:
I will go ahead and close this ticket since the proposed heuristic has been implemented in #323. Please feel free to continue the conversation here and the heuristic can be updated if desired.

closed 2018-11-15 by dirk-thomas

jacobperron, 2018-11-15:
> Can you point to an existing message file which uses this style?

Nope