ros2/rosidl #342: [rosidl_parser] Possible incorrect grammar for 'scoped_name'

jacobperron, 2018-12-21:
## Bug report

Follow up from ros2/ci#231.

**Required Info:**

- Operating System:
  - All
- Installation type:
  - source
- Version or commit hash:
  - 43eee85d1b855f5f1969d598995f0548cac1a1e7
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

For some reason, the latest changes on [lark@0.7b](https://github.com/lark-parser/lark/commits/0.7b) seem to have uncovered a bug in the IDL grammar.

1. Create a simple message definition, `MyMessage.idl`:

```
struct MyMessage {
  float float_value;
};
```

2. Copy the [IDL grammar](https://github.com/ros2/rosidl/blob/9535733cd9edb2aab0d485fee16e660eb68c5e2c/rosidl_parser/rosidl_parser/grammar.lark) to the same directory.

3. Install the latest changes for lark@0.7b:

```bash
sudo -H python3 -m pip install -U git+https://github.com/lark-parser/lark.git@0.7b
```
4. Run the following Python script and note the output:

*test.py*
```python
import os
from lark import Lark

def main():
    grammar_file = os.path.join(os.path.dirname(__file__), 'grammar.lark')
    with open(grammar_file, mode='r', encoding='utf-8') as h:
        grammar = h.read()

    parser = Lark(grammar, start='specification')

    input_file = os.path.join(os.path.dirname(__file__), 'MyMessage.idl')
    with open(input_file, mode='r', encoding='utf-8') as h:
        input_string = h.read()

    tree = parser.parse(input_string)
    struct_def = list(tree.find_data('struct_def'))
    print(struct_def[0])

if __name__ == '__main__':
    main()
```

E.g.
```bash
$ python3 test.py
Tree(struct_def, [Token(IDENTIFIER, 'MyMessage'), Tree(member, [Tree(type_spec, [Tree(simple_type_spec, [Tree(base_type_spec, [Tree(floating_pt_type, [Tree(simple_type_spec, [Tree(scoped_name, [Token(IDENTIFIER, 'float')])])])])])]), Tree(declarators, [Tree(declarator, [Tree(simple_declarator, [Token(IDENTIFIER, 'float_value')])])])])])
```

#### Expected behavior

To see what (I think) the tree structure should look like, revert the version of lark installed and run the script again:

```bash
$ sudo -H python3 -m pip install -U git+https://github.com/lark-parser/lark.git@8415fa26a3d3d2b81e64e4fe440faab15b53db49
$ python3 test.py
Tree(struct_def, [Token(IDENTIFIER, 'MyMessage'), Tree(member, [Tree(type_spec, [Tree(simple_type_spec, [Tree(base_type_spec, [Tree(floating_pt_type, [Tree(floating_pt_type_float, [])])])])]), Tree(declarators, [Tree(declarator, [Tree(simple_declarator, [Token(IDENTIFIER, 'float_value')])])])])])
```

#### Actual behavior

Here's the diff between the two outputs:

```diff
diff --git a/expected.out b/actual.out
index f657b6f..45d5e80 100644
--- a/expected.out
+++ b/actual.out
@@ -1 +1 @@
-Tree(struct_def, [Token(IDENTIFIER, 'MyMessage'), Tree(member, [Tree(type_spec, [Tree(simple_type_spec, [Tree(base_type_spec, [Tree(floating_pt_type, [Tree(floating_pt_type_float, [])])])])]), Tree(declarators, [Tree(declarator, [Tree(simple_declarator, [Token(IDENTIFIER, 'float_value')])])])])])
+Tree(struct_def, [Token(IDENTIFIER, 'MyMessage'), Tree(member, [Tree(type_spec, [Tree(simple_type_spec, [Tree(base_type_spec, [Tree(floating_pt_type, [Tree(simple_type_spec, [Tree(scoped_name, [Token(IDENTIFIER, 'float')])])])])])]), Tree(declarators, [Tree(declarator, [Tree(simple_declarator, [Token(IDENTIFIER, 'float_value')])])])])])
```
#### Additional information

The `scoped_name` element in the output seemed suspicious to me as there are no scoped names in the input message. So I took a closer look at the grammar and thing the following rule should change:

```diff
diff --git a/grammar.lark b/update_grammar.lark
index 37e8214..4dfdf77 100644
--- a/grammar.lark
+++ b/update_grammar.lark
@@ -106,8 +106,7 @@ definition: module_dcl ";"
 module_dcl: annotation_appl* "module" IDENTIFIER "{" definition+ "}"
 
 // (4)
-scoped_name.1: IDENTIFIER
-  | scoped_name_separator IDENTIFIER
+scoped_name.1: scoped_name_separator IDENTIFIER
   | scoped_name scoped_name_separator IDENTIFIER
 // separate rule to identify the separator
 scoped_name_separator: "::"
```

I'm not 100% sure, but I don't think a scoped name is intended to **not** contain a `::`.

A PR will follow after I confirm this resolves the original failing tests (https://ci.ros2.org/job/ci_linux/5926/testReport/).

jacobperron, 2018-12-21:
After consulting the IDL spec, believe the original scoped_name rule is correct. It is possible this is an issue with the change in lark.

jacobperron, 2018-12-21:
There is also non-determinism in the output. Running the example with the latest version of lark@0.7b *sometimes* produces the expected output (about 50% of the time). Reverting to an earlier commit of lark does not show this non-determinism. Sounds related to lark-parser/lark#278.

dirk-thomas, 2019-01-14:
See https://github.com/lark-parser/lark/issues/278#issuecomment-449455205

Updated instructions:
* ros2/ci#238
* ros2/ros2_documentation#96

closed 2019-01-14 by dirk-thomas