ros2/rosidl #344: Namespace msg types failing for actions

AustinDeric, 2018-12-30:
Using namespaced message types such as `geometry_msgs/Pose pose` in actions fails.  

I created a simple example using the example_interfaces so you can reproduce:
https://github.com/AustinDeric/example_interfaces/tree/action-namespace-failing

In this example:
the .action file is:
```
# Goal
int32 order
geometry_msgs/Pose pose
---
# Result
int32[] sequence
---
# Feedback
int32[] sequence
```

The CMakeLists is:
```
cmake_minimum_required(VERSION 3.5)

project(example_interfaces)

# Default to C++14
if(NOT CMAKE_CXX_STANDARD)
  set(CMAKE_CXX_STANDARD 14)
endif()
if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

find_package(action_msgs REQUIRED)
find_package(ament_cmake REQUIRED)
find_package(rosidl_default_generators REQUIRED)
find_package(geometry_msgs REQUIRED)

rosidl_generate_interfaces(${PROJECT_NAME}
  "action/Fibonacci.action"
  "srv/AddTwoInts.srv"
  DEPENDENCIES action_msgs geometry_msgs
)

install(FILES mapping_rules.yaml DESTINATION share/${PROJECT_NAME})

ament_export_dependencies(rosidl_default_runtime)
ament_package()
```

The build error is:
```
$ colcon build --symlink-install
Starting >>> example_interfaces
--- stderr: example_interfaces                               
Action interface 'geometry_msgs/PointStamped' contains an unknown field type: geometry_msgs/PointStamped target
make[2]: *** [rosidl_typesupport_cpp/example_interfaces/action/fibonacci__type_support.cpp] Error 1
make[1]: *** [CMakeFiles/example_interfaces__rosidl_typesupport_cpp__generate_actions.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
---
Failed   <<< example_interfaces	[ Exited with code 2 ]

Summary: 0 packages finished [19.7s]
  1 package failed: example_interfaces
  1 package had stderr output: example_interfaces
```

hidmic, 2019-01-03:
I cannot reproduce with a fresh source build on Ubuntu 18.04. @AustinDeric would you mind sharing your setup (platform, installation type, etc., as described [here](https://index.ros.org/doc/ros2/Developer-Guide/#issues))?

AustinDeric, 2019-01-03:
OS: 16.04
ROS2: Crystal from source
DDS: Default RMW implementation

I'll send you a CI pipline log & docker image to help reproduce. Thanks for the feedback!

hidmic, 2019-01-07:
@AustinDeric I cannot reproduce on Ubuntu 16.04 either.

ibaiape, 2019-01-31:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04
- Installation type:
  - From source
- Version or commit hash:
  - Crystal [unpatched](https://raw.githubusercontent.com/ros2/ros2/release-latest/ros2.repos)(as of December 19) and [patch 1](https://raw.githubusercontent.com/ros2/ros2/crystal-patch-1/ros2.repos) hash 4659eeb1aa8f5529ad9b76e568318dfcb0390c03 (same error output with both).
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
Clean installation:
```
mkdir -p ~/ros2_ws/src
cd ~/ros2_ws
wget https://raw.githubusercontent.com/ros2/ros2/release-latest/ros2.repos
    or
wget https://raw.githubusercontent.com/ros2/ros2/crystal-patch-1/ros2.repos
vcs import src < ros2.repos

sudo rosdep init
rosdep update
rosdep install --from-paths src --ignore-src --rosdistro crystal -y --skip-keys "console_bridge fastcdr fastrtps libopensplice67 libopensplice69 python3-lark-parser rti-connext-dds-5.3.1 urdfdom_headers"

colcon build --symlink-install --packages-ignore qt_gui_cpp rqt_gui_cpp
```
Reproduce error:
```
source {ros2_crystal_ws}/install/setup.bash

git clone https://github.com/AustinDeric/example_interfaces/ -b action-namespace-failing

cd example_interfaces

colcon build
```
#### Expected behavior
Package finishes compilation without errors.
#### Actual behavior
Compilation fails with stderr (spanish included...):
```
--- stderr: example_interfaces                               
Action interface 'geometry_msgs/Pose' contains an unknown field type: geometry_msgs/Pose pose
make[2]: *** [rosidl_typesupport_cpp/example_interfaces/action/fibonacci__type_support.cpp] Error 1
make[1]: *** [CMakeFiles/example_interfaces__rosidl_typesupport_cpp__generate_actions.dir/all] Error 2
make[1]: *** Se espera a que terminen otras tareas....
make: *** [all] Error 2
---
```
#### Additional information
Equal error from trying to include custom message structures (custom .msg instead of from geometry_msgs).
Equal error from compiling the included message in the same package or separately and setting dependencies.

I'm new to this bug reporting thing, so apologies in advance for any missing information. Anything else I can provide, just ask.

ibaiape, 2019-01-31:
So the exception that comes out originates from src/ros2/rosidl/rosidl_adapter/rosidl_adapter/parse.py at line 805:

```
800  for field in fields:
801       if field.type.is_primitive_type():
802           continue
803       base_type = BaseType(BaseType.__str__(field.type))
804       if base_type not in known_msg_types:
805           raise UnknownMessageType(
806               "%s interface '%s' contains an unknown field type: %s" %
807               (spec_type, base_type, field))
```

Seeing I can modify this to get some output, I did some basic tinkering to print the `known_msg_types` list and see what's happening:
```
knownList = "[\n"
for msg in known_msg_types:
    knownList += "  " +  str(msg) + "\n"
knownList += "]"
if base_type not in known_msg_types:
    raise UnknownMessageType(
        "%s interface '%s' contains an unknown field type: %s, known files contain: %s" %
        (spec_type, base_type, field, knownList))
```
Which yields:
```
Action interface 'geometry_msgs/Pose' contains an unknown field type: geometry_msgs/Pose pose, known files contain: [
  action_msgs/GoalInfo
  action_msgs/GoalStatus
  action_msgs/GoalStatusArray
  action_msgs/CancelGoal_Request
  action_msgs/CancelGoal_Response
  builtin_interfaces/Duration
  builtin_interfaces/Time
  unique_identifier_msgs/UUID
]

```

Adding a .msg file without issues (using builtin_interfaces/Duration) to the CMakeLists.txt before the action and forcing an exception by pasting the raise out of the check like so:
```
knownList = "[\n"
for msg in known_msg_types:
    knownList += "  " +  str(msg) + "\n"
knownList += "]"
raise UnknownMessageType(
    "%s interface '%s' contains an unknown field type: %s, known files contain: %s" %
    (spec_type, base_type, field, knownList))
if base_type not in known_msg_types:
    raise UnknownMessageType(
        "%s interface '%s' contains an unknown field type: %s, known files contain: %s" %
        (spec_type, base_type, field, knownList))
```

Outputs:
```
Service interface 'unique_identifier_msgs/UUID' contains an unknown field type: unique_identifier_msgs/UUID action_goal_id, known files contain: [
  example_interfaces/Duration
  example_interfaces/AddTwoInts_Request
  example_interfaces/AddTwoInts_Response
  example_interfaces/Fibonacci_Goal_Request
  example_interfaces/Fibonacci_Goal_Response
  example_interfaces/Fibonacci_Result_Request
  example_interfaces/Fibonacci_Result_Response
  example_interfaces/Fibonacci_Feedback
  action_msgs/GoalInfo
  action_msgs/GoalStatus
  action_msgs/GoalStatusArray
  action_msgs/CancelGoal_Request
  action_msgs/CancelGoal_Response
  builtin_interfaces/Duration
  builtin_interfaces/Time
  unique_identifier_msgs/UUID
  geometry_msgs/Accel
  geometry_msgs/AccelStamped
  geometry_msgs/AccelWithCovariance
  geometry_msgs/AccelWithCovarianceStamped
  geometry_msgs/Inertia
  geometry_msgs/InertiaStamped
  geometry_msgs/Point
  geometry_msgs/Point32
  geometry_msgs/PointStamped
  geometry_msgs/Polygon
  geometry_msgs/PolygonStamped
  geometry_msgs/Pose
  geometry_msgs/Pose2D
  geometry_msgs/PoseArray
  geometry_msgs/PoseStamped
  geometry_msgs/PoseWithCovariance
  geometry_msgs/PoseWithCovarianceStamped
  geometry_msgs/Quaternion
  geometry_msgs/QuaternionStamped
  geometry_msgs/Transform
  geometry_msgs/TransformStamped
  geometry_msgs/Twist
  geometry_msgs/TwistStamped
  geometry_msgs/TwistWithCovariance
  geometry_msgs/TwistWithCovarianceStamped
  geometry_msgs/Vector3
  geometry_msgs/Vector3Stamped
  geometry_msgs/Wrench
  geometry_msgs/WrenchStamped
  std_msgs/Bool
  std_msgs/Byte
  std_msgs/ByteMultiArray
  std_msgs/Char
  std_msgs/ColorRGBA
  std_msgs/Empty
  std_msgs/Float32
  std_msgs/Float32MultiArray
  std_msgs/Float64
  std_msgs/Float64MultiArray
  std_msgs/Header
  std_msgs/Int16
  std_msgs/Int16MultiArray
  std_msgs/Int32
  std_msgs/Int32MultiArray
  std_msgs/Int64
  std_msgs/Int64MultiArray
  std_msgs/Int8
  std_msgs/Int8MultiArray
  std_msgs/MultiArrayDimension
  std_msgs/MultiArrayLayout
  std_msgs/String
  std_msgs/UInt16
  std_msgs/UInt16MultiArray
  std_msgs/UInt32
  std_msgs/UInt32MultiArray
  std_msgs/UInt64
  std_msgs/UInt64MultiArray
  std_msgs/UInt8
  std_msgs/UInt8MultiArray
]
```

I added a .msg file as forcing an exception with the .action shows comparable results as before, although the compilation breaks up at `unique_identifier_msgs/UUID`, assuming that's because it always gets through the process first. The rest of the CMakeLists.txt and package.xml was unchanged outside of adding Duration.msg to the file list (based on the repo referenced beforehand).

It seems like actions aren't taking dependencies declared at the CMakeLists.txt into account. I might be wrong on this, but delving further into the innards of ROS 2 is out of my current capabilities.

Hope this can be of use, do ask for any information that might be of help.

dirk-thomas, 2019-01-31:
Your example package builds just fine for me too - on Ubuntu 18.04 using the latest Debian packages of Crystal.

Is there anything non-standard/default with your system?

ibaiape, 2019-02-01:
> @dirk-thomas 
> Is there anything non-standard/default with your system?

Not that I know of.

Just in case I gave a fresh Ubuntu 18.04 a go (in a VM). Installed ros2 crystal through the debian packages too. Got the same error.

Are you by any chance compiling the master branch of the referenced repo? **The modifications are at the `action-namespace-failing`**, the master branch is equal to the current official one (there's no non-primitives property in the action).

ibaiape, 2019-02-01:
Managed to compile manually tinkering with build files. After some digging I found out the "known_msg_files" is taken as an argument, and that those files are stored in a .json when building. So I took what looked like the full list and pasted it in what looked like the one the action generator was using.

The process I followed:
```
git clone https://github.com/AustinDeric/example_interfaces/ -b action-namespace-failing
cd example_interfaces
colcon build
```
Got the error output mentioned above, then:

I copied the `ros_interface_dependencies` and `target_dependencies` lists from:
```
example_interfaces/build/example_interfaces/rosidl_typesupport_cpp__arguments.json
```
And substituted the ones in:
```
example_interfaces/build/example_interfaces/rosidl_typesupport_cpp__generate_actions__arguments.json
```

Then ran the build again.
```
colcon build
```

And it compiles. I can source the installation just fine but I haven't tried the action in the examples just yet.

I also tried this with some custom packages&actions and the hack seems to work. Same as with this one, the compilation ends and sourcing the installation doesn't spit out any error.

Will test the actions and return with more info.

ibaiape, 2019-02-01:
Modifying the examples (minimal_action_client & server) to pass & receive the modified actions works, I managed to pass & print values inside the Pose structure.

ibaiape, 2019-02-01:
It seems the dependencies get properly linked for the generation of the .msg&.srv files the action interfaces depend upon:

**rosidl_typesupport_cpp__arguments.json**:
```
"ros_interface_files": [
  "{action}_Goal.srv",
  "{action}_Goal_Request.msg",
  "{action}_Goal_Response.msg",
  "{action}_Result.srv",
  "{action}_Result_Request.msg",
  "{action}_Result_Response.msg",
  "{action}_Feedback.msg"
]
```
But aren't passed onto the one that compiles the action files themselves:

**rosidl_typesupport_cpp__generate_actions__arguments.json**
```
"ros_interface_files": [
  "{action}.action"
]
```
Looks like `rosidl_typesupport_cpp__arguments.json` is the one used to generate both .srvs and .msgs across all packages. Might it simply be that those dependencies aren't "redirected" or "copied over" for the action generator?

dirk-thomas, 2019-02-05:
It looks like this is causes by a bug which has been fixed in #343.

Please try the patch from that branch and comment if that resolved the problem for you.

ibaiape, 2019-02-05:
It does! Both the linked repository and custom packages.

Much appreciated! Thanks for the quick support!

FYI it **works in a patch 1 installation**, but doesn't seem to do the trick in a "day 1" ros2 crystal install.

dirk-thomas, 2019-02-05:
> FYI it **works in a patch 1 installation**, but doesn't seem to do the trick in a "day 1" ros2 crystal install.

I am very surprised to hear that because the referenced patch has not been released yet and is not part of the first patch release.

ibaiape, 2019-02-06:
Actually, my bad, it does work.

I remembered to re-build my patch 1 rosidl_cmake but I didn't on the day 1 installation, rookie mistake.

Thanks again for pointing out the fix!

dirk-thomas, 2019-02-06:
With the already merged PR addressing the problem I will go ahead and close this ticket. The patch should be released into Crystal in the next patch release.

closed 2019-02-06 by dirk-thomas