ros2/rosidl #345: Wrong type mapping? (differs from osidl_generator_dds_idl)

TSC21, 2019-01-18:
## Bug report

I am using ROS2 Crystal `rosidl_adapter` to generate the IDL files for Fast RTPS with the usage of `rosidl_generate_interfaces()` CMake directive. The current issue is that the generated IDL files come with types that are unknown to `fastrtpsgen` to generate `PubSubTypes`.

The current mapping on `rosidl_adapter` is https://github.com/ros2/rosidl/blob/master/rosidl_adapter/rosidl_adapter/msg/__init__.py#L43-L58, but it is expected something as https://github.com/PX4/Firmware/blob/master/msg/tools/px_generate_uorb_topic_helper.py#L78-L93. `rosidl_generator_dds_idl` does us the same mapping and it works as expected: https://github.com/ros2/rosidl_dds/blob/master/rosidl_generator_dds_idl/rosidl_generator_dds_idl/__init__.py#L135-L150.

So my question is: why is the mapping different on this case? Cause the one in `rosidl_adapter` does not seem to be functioning properly.

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Binary install of ROS2 Crystal packages
- Version or commit hash:
  - Crystal release
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
Use CMake `rosidl_generate_interfaces()` to generate IDL files and try `/fastrtpsgen -example x64Linux2.6gcc` over the generated IDL files will result in errors like:

```sh
Processing the file /home/nuno/PX4/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg/QshellRetval.idl...
/home/nuno/PX4/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg/QshellRetval.idl:11:6: error: 'uint64' was not defined previously
/home/nuno/PX4/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg/QshellRetval.idl:13:6: error: 'int32' was not defined previously
/home/nuno/PX4/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg/QshellRetval.idl:15:13: error: mismatched input 'sequence' expecting {'::', ID}
```

#### Expected behavior
It's expected that `fastrtpsgen` generates the `PubSubTypes` correctly without throwing errors on unknown types.

#### Actual behavior
`fastrpsgen` fails on building against IDL types generated from `rosidl_adapter`.

@dirk-thomas any input here? Thanks!

dirk-thomas, 2019-01-18:
The mapping in `rosidl_adapter` uses the types specified in IDL 4.2. In that version of the specification all the integer types are available with their bit length as a suffix - `(u)intN`. Earlier versions of the spec are only defining named types like `(unsigned)` `short`, `long`, `long long` (and don't have an 8-bit type).

`fastrtpsgen` might simply not yet support these type names. Since ROS 2 doesn't use `fastrtpsgen` we didn't run into that problem with FastRTPS. For Connext and OpenSplice we generate slightly modified `.idl` files using the old integer types since the current versions don't support the new integer type names either. Upcoming versions from the DDS implementations are expected to support the new integer type names though.

TSC21, 2019-01-18:
Ok so based on the above, should I just drop the usage of the IDL messages generated by `rosidl_apdater` and keep using the ones generated by `rosidl_generator_dds_idl()`? Would it make sense to support on `rosidl_adapter` loading a mapping depending on the `rmw_implementation` being used on?

dirk-thomas, 2019-01-18:
It would probably best to fill a ticket for `fastrtpsgen` to get updated to support all types defined in the current standard.

> should I just drop the usage of the IDL messages generated by `rosidl_apdater` and keep using the ones generated by `rosidl_generator_dds_idl()`?

Yes, you can use the output `rosidl_generator_dds_idl()` as a workaround until the tool gets updated.

> Would it make sense to support on `rosidl_adapter` loading a mapping depending on the `rmw_implementation` being used on?

That is not feasible since you can build ROS 2 with multiple RMW implementations at the same time so `rosidl_adapter` can generate "custom" files.

TSC21, 2019-01-18:
Alright thanks! I will do what you suggest.

Also along this topic: if one generates the ROS messages in one package and then use the generated ROS messages to generate the IDLs in another packages using `rosidl_generator_dds_idl()`, should I expect it to work? Cause I have been trying that and similarly to what was happening before, it doesn't seem to generate them. And I am making sure that I am creating a dependency on a target using `add_dependencies(${PROJECT_NAME}__fastrtps_idl micrortps_agent)`.
(Just brought this in here as it doesn't seem a question to open a new issue).
Thanks once again!

dirk-thomas, 2019-01-18:
> Also along this topic: if ...

I am sorry but I can't follow the scenario without more context / information.

TSC21, 2019-01-18:
Alright I am going to open a new issue regarding this then. Thanks!

closed 2019-01-18 by TSC21

dirk-thomas, 2019-01-21:
The current code in ROS 2 follows the latest version of the IDL specification which is version 4.2. this is mentioned in the design doc: https://github.com/ros2/design/blob/97d581287b77300668493aa9e525cea9f0dd7c24/articles/142_idl.md