ros2/rosidl #348: Actions with self-defined interfaces fail compilation

ibaiape, 2019-02-12:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04
- Installation type:
  - From source
- Version or commit hash:
  - Crystal patch 1 (plus https://github.com/ros2/rosidl/pull/343/files)
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
```
source {ros2_crystal_ws}/install/setup.bash

git clone https://github.com/ibaiape/example_interfaces
cd example_interfaces
colcon build
```

#### Expected behavior

Package finishes compilation without errors.

#### Actual behavior
Compilation fails with stderr (spanish included...):
```
--- stderr: example_interfaces                               
Action interface 'example_interfaces/ExampleMessage' contains an unknown field type: example_interfaces/ExampleMessage example
make[2]: *** [rosidl_typesupport_cpp/example_interfaces/action/fibonacci__type_support.cpp] Error 1
make[1]: *** [CMakeFiles/example_interfaces__rosidl_typesupport_cpp__generate_actions.dir/all] Error 2
make[1]: *** Se espera a que terminen otras tareas....
make: *** [all] Error 2
---
```

#### Additional information

If a package tries to compile an action that includes an interface defined in the package itself, the compilation fails.

Related to https://github.com/ros2/rosidl/issues/344 (at least feels like it is).

The patch referenced there fixes compilation when the included interface comes from another package (so it's already compiled and passed as a dependency). However, the same error comes out when said interface is defined in the same package the action is (the problem described here).

It seems when compiling actions the files defined in the same package don't get referenced for the action generator, so it doesn't know they exist.

dirk-thomas, 2019-02-12:
Have you tried to reference the message type without the package name as a workaround? Instead of:

`example_interfaces/ExampleMessage example`

trying:

`ExampleMessage example`

ibaiape, 2019-02-12:
I get the same output, including the:
```
Action interface 'example_interfaces/ExampleMessage' contains an unknown field type: example_interfaces/ExampleMessage example
```
line, assuming it prepends it's own package if it doesn't detect it elsewhere.

dirk-thomas, 2019-02-12:
Can you try the same with the upcoming 2nd patch release of Crystal which actually contains the referenced patch (see ros2/ros2#655)? Since you are using the 1st patch release you don't have the fix from #343.

dirk-thomas, 2019-02-12:
Nvm I guess you build that from source:

> Crystal patch 1 (plus https://github.com/ros2/rosidl/pull/343/files)

ibaiape, 2019-02-12:
I have compiled the current crystal master (commit https://github.com/ros2/ros2/commit/a76d265cd57d1b0a3dd98dee8bd85ceac1769a05) to the same result.

Same when not specifying the package.

sloretz, 2019-02-12:
I'm able to reproduce this on master.

The error comes from [`rosidl_apapter:validate_field_types()`](https://github.com/ros2/rosidl/blob/7c5c7d3af98fb8f5fd8ae9112ee278d21f899df1/rosidl_adapter/rosidl_adapter/parser.py#L806) which is given a list of [`known_msg_types`](https://github.com/ros2/rosidl/blob/7c5c7d3af98fb8f5fd8ae9112ee278d21f899df1/rosidl_adapter/rosidl_adapter/parser.py#L785). Known message types come from two places in the generator arguments: [interface files being generated, and other packages interfaces](https://github.com/ros2/rosidl_typesupport/blob/bda9bb44c5bdd1a0539e2a0867c14093d4d117ef/rosidl_typesupport_cpp/rosidl_typesupport_cpp/__init__.py#L55-L56). Services can find messages in the same packge because they're generated at the same time, so both are added to `known_msg_types` by the generator argument [`ros_interface_files`](https://github.com/ros2/rosidl_typesupport/blob/bda9bb44c5bdd1a0539e2a0867c14093d4d117ef/rosidl_typesupport_cpp/cmake/rosidl_typesupport_cpp_generate_interfaces.cmake#L76). The services and messages that make up an action are generated at the same time as regular messages and services, so they are actually generated correctly too. The overall action [is generated separately](https://github.com/ros2/rosidl/blob/7c5c7d3af98fb8f5fd8ae9112ee278d21f899df1/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake#L300-L306), so it does not know about field types in the same package because `ros_interface_files` only contains the action interface files.

@dirk-thomas I'm assuming this won't be an issue once the IDL refactor lands because everything will be generated in the same step. For crystal I think the simplest workaround is to [comment out this call to `validate_field_types()`](https://github.com/ros2/rosidl_typesupport/blob/bda9bb44c5bdd1a0539e2a0867c14093d4d117ef/rosidl_typesupport_cpp/rosidl_typesupport_cpp/__init__.py#L96). While it means the action fields aren't validated when the overall action is generated, each service and message that makes up an action gets their fields validated when they are generated.

dirk-thomas, 2019-02-12:
> I'm assuming this won't be an issue once the IDL refactor lands because everything will be generated in the same step.

Correct.

closed 2019-02-25 by sloretz