ros2/rosidl #349: Should the <msg>__struct.hpp setters (set__<param>) return a reference instead of a pointer?

cho3, 2019-02-12:
## Feature request

#### Feature description

ROS 2's generated messages claim to follow the [Named Parameter Idiom](https://isocpp.org/wiki/faq/ctors#named-parameter-idiom), e.g:

```cpp
File f = OpenFile("foo.txt")
           .readonly()
           .createIfNotExist()
           .appendWhenWriting()
           .blockSize(1024)
           .unbuffered()
           .exclusiveAccess();
```
The canonical presentation of which uses references, e.g.:

```cpp
class OpenFile {
public:
  OpenFile(const std::string& filename);
    // sets all the default values for each data member
  OpenFile& readonly();  // changes readonly_ to true
  OpenFile& readwrite(); // changes readonly_ to false
  OpenFile& createIfNotExist();
  OpenFile& blockSize(unsigned nbytes);
};
```
However, [the generated code](https://github.com/ros2/rosidl/blob/master/rosidl_generator_cpp/resource/msg__struct.hpp.em#L215..L220) returns a pointer, e.g:

```cpp
  Type * set__@(field.name)(
    const @(msg_type_to_cpp(field.type)) & _arg)
  {
    this->@(field.name) = _arg;
    return this;
  }
```

This results in somewhat awkward, and possibly non-canonical method chaining, e.g:

```cpp
void foo(const geometry_msgs::msg::Point32 & pt);

foo(Point32().set__x(1)->set__y(2)->set__z(3));
```

Should the return type be a reference instead, e.g:

```cpp
  Type & set__@(field.name)(
    const @(msg_type_to_cpp(field.type)) & _arg)
  {
    this->@(field.name) = _arg;
    return *this;
  }
```

Resulting in the following usage:

```cpp
void foo(const geometry_msgs::msg::Point32 & pt);

foo(Point32().set__x(1).set__y(2).set__z(3));
```

Or was there a reason why a pointer was returned instead?

wjwwood, 2019-02-15:
Returning a reference would make more sense to me, but there may be a historical reason for it of which I'm unaware.

@ros2/team anyone know why this wouldn't be a good change to make?

Assuming no one has an objection, we'd appreciate help making the change with a pr.

wjwwood, 2019-02-15:
One reason for it may be that we often instantiate the message instances as a shared or unique ptr, in which case you'd see something more like this:

```c++
auto msg = std::make_shared<geometry_msgs::msg::Vector>();
pub->publish(msg->set_x(1)->set_y(1)->set_z(1));
```

Which appears more natural at first glance.

dirk-thomas, 2019-03-07:
> One reason for it may be that we often instantiate the message instances as a shared or unique ptr, ...

Based on that comment do we actually want to change the API to return references or keep pointers since smart pointers are the more common usage? If it is the latter I would suggest to close the ticket rather than keeping it open with "help wanted".

wjwwood, 2019-03-12:
We can close it and reopen it if others have more feedback.

closed 2019-03-12 by wjwwood

dirk-thomas, 2019-03-12:
See #353 for a patch implementing the proposed change.