ros2/rosidl #351: Support floating point special values

johnmarkwayve, 2019-03-04:
## Feature request

Right now, it appears that the rosidl will only accept floating point values specified in decimal or exponential form. However, there are several valid special values that can't be specified this way. In particular:

- NaN
- +/- infinity
- It's not clear whether or not signed zero values are supported

#### Feature description

It would be good to support all floating point special values. In particular, infinities and NaNs can be useful as default values or constants in certain situations.

#### Implementation considerations

Simply support specification of float32/float64 values (constant and default) by some special strings. e.g.

- "nan", "+nan" for positive nan
- "-nan" for negative nan
- "inf", "+inf", "-inf" similarly
- "+0", "-0"

Ensure use of these cases are tested and documented.

bijoua29, 2022-09-12:
I think this is the same problem I am seeing when echoing a topic with NaNs.

ros2 topic echo /controller/state
make_tuple(): unable to convert arguments to Python object (compile in debug mode for details)

Is there anyone working on this issue? I am willing to try if someone can point me to where the issue might be.

oysstu, 2022-09-13:
> I think this is the same problem I am seeing when echoing a topic with NaNs.

This issue refers to specifying constants and default values in the idl. I think what you are referring to is the issue in https://github.com/ros2/rosidl_python/issues/162. It's currently fixed in rolling, but I don't know if it's going to be backported to humble.

A quick workaround is either running python with -O or exporting PYTHONOPTIMIZE=1 as an environmental variable, that should disable the NaN/Inf assertions.

bijoua29, 2022-09-15:
@oysstu I am on the latest rolling (from a few days ago) and still get an error but the message is slightly different:  'Unable to convert call argument to Python object (compile in debug mode for details)'. Maybe the fix didn't make it into the recent sync. Which package does this reside in so I can check the version I have? 

Setting PYTHONOPTIMIZE=1 does fix the problem. I can work with that for the time being.

clalancette, 2022-09-15:
> Maybe the fix didn't make it into the recent sync.

I just released it yesterday, so it will be in the next sync.

bijoua29, 2022-09-15:
@clalancette Ah perfect. Thanks for following up.