ros2/rosidl #352: support for WString across all generators

dirk-thomas, 2019-03-12:
Follow up of ros2/rosidl#346.

dirk-thomas, 2019-05-02:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

jacobperron, 2019-05-02:
Looks like you've addressed all the warnings/failures. Considering you are adding more work on top of this, we can merge as-is and catch anything we missed in the next round of CI.

dirk-thomas, 2019-05-02:
I went over all PRs again and resolved the remaining comments. In favor of time I will merge the changes now and trigger a set of CI builds afterwards to double check. :crossed_fingers:

closed 2019-05-02 by dirk-thomas