ros2/rosidl #354: Building interfaces without Python support

alsora, 2019-03-19:
## Bug report

Before this PR https://github.com/ros2/rosidl/issues/346, which introduced parsing of IDL format, I was able to build ROS2 without Python support, for platforms which do not have it installed.

I was basically following the instructions in https://github.com/ros2-for-arm/ros2/issues/5#issuecomment-402676698 updated to the current packages structure

```
sed -i -r \
    's/<build(.+?py.+?)/<\!\-\-build\1\-\->/' \
    src/ros2/rosidl_defaults/rosidl_default_generators/package.xml
```

Unfortunately it looks like this is not working anymore since the latest updates.

When I try to build the ROS2 SDK, the cross-compilation fails as soon as I reach some the `builtin_interfaces`

```
Starting >>> builtin_interfaces
--- stderr: builtin_interfaces                                   
Traceback (most recent call last):
  File "/root/ws/install/share/rosidl_generator_cpp/cmake/../../../lib/rosidl_generator_cpp/rosidl_generator_cpp", line 8, in <module>
    from rosidl_generator_cpp import generate_cpp
  File "/root/ws/install/lib/python3.6/site-packages/rosidl_generator_cpp/__init__.py", line 17, in <module>
    from rosidl_cmake import generate_files
  File "/root/ws/install/lib/python3.6/site-packages/rosidl_cmake/__init__.py", line 27, in <module>
    from rosidl_parser.parser import parse_idl_file
  File "/root/ws/install/lib/python3.6/site-packages/rosidl_parser/parser.py", line 18, in <module>
    from lark import Lark
ModuleNotFoundError: No module named 'lark'
make[2]: *** [rosidl_generator_cpp/builtin_interfaces/msg/duration.hpp] Error 1
make[1]: *** [CMakeFiles/builtin_interfaces__cpp.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Traceback (most recent call last):
  File "/root/ws/install/share/rosidl_generator_c/cmake/../../../lib/rosidl_generator_c/rosidl_generator_c", line 8, in <module>
    from rosidl_generator_c import generate_c
  File "/root/ws/install/lib/python3.6/site-packages/rosidl_generator_c/__init__.py", line 15, in <module>
    from rosidl_cmake import convert_camel_case_to_lower_case_underscore
  File "/root/ws/install/lib/python3.6/site-packages/rosidl_cmake/__init__.py", line 27, in <module>
    from rosidl_parser.parser import parse_idl_file
  File "/root/ws/install/lib/python3.6/site-packages/rosidl_parser/parser.py", line 18, in <module>
    from lark import Lark
ModuleNotFoundError: No module named 'lark'
make[2]: *** [rosidl_generator_c/builtin_interfaces/msg/duration.h] Error 1
make[1]: *** [CMakeFiles/builtin_interfaces__rosidl_generator_c.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< builtin_interfaces	[ Exited with code 2 ]
```

`lark` is obviously not present in my sysroot, but, before this change, I was able to build everything without using the `rosidl_parser/parser.py` script.

In order to make the cross-compilation of the ros2 master succeed, I had to revert the following packages to the commit previous to the aforementioned PR

 - `rosidl`
 - `rosidl_typesupport`
 - `rosidl_typesupport_fastrtps`

Is it possible with the new update to keep building ROS2 without using `lark` or other python dependencies?

Thank you

dirk-thomas, 2019-03-19:
`lark` is now being used to parse the `.idl` files. So it is not possible to remove that dependency.

sloretz, 2019-03-19:
> Before this PR #346, which introduced parsing of IDL format, I was able to build ROS2 without Python support, for platforms which do not have it installed.

> In order to make the cross-compilation of the ros2 master succeed, I had to revert the following packages to the commit previous to the aforementioned PR

Are you cross-compiling for a target system that cannot run python, or cross-compiling on a build machine that does not have python installed?

alsora, 2019-03-20:
@sloretz  sorry it was not clear.

I'm cross-compiling for a target system that cannot run python

sloretz, 2019-03-20:
> I'm cross-compiling for a target system that cannot run python

It seems like this should be possible. `Lark` should only be needed on the machine doing the building, not on the target. I'm not familiar with cross compiling ROS 2 though.

Are you aware of https://index.ros.org/doc/ros2/Tutorials/Cross-compilation/ ? If that tutorial is different from the instructions you're following, does the tutorial work for you?

alsora, 2019-03-20:
I'm doing something very similar to the cross-compilation tutorial.
Actually `lark` is not used at all in that tutorial, neither in the build or the target environment.

The only difference I have is that my sysroot only contains a very minimal set of dependencies.

I installed `lark` on the machine doing the build, but I get the same error.

Should we move this discussion to cross-compilation?

alsora, 2019-03-22:
@sloretz you were right, with `lark` installed on the machine doing the build I'm able to cross-compile.

Thank you.

We can close this.

closed 2019-03-22 by alsora

Ryanf55, 2024-08-01:
Is there a way to disable the python generator in CMake? 

Building px4_msgs takes over an hour on CI.