ros2/rosidl #356: Fatal Error when generating actions for opensplice

scgroot, 2019-03-25:
## Bug report
When generating actions for opensplice I get the following error:
```
In file included from /home/user/proj/git/my_ws/build/my_msgs/rosidl_typesupport_opensplice_cpp/my_msgs/action/dds_opensplice/ccpp_MyAction_.h:4:0,
                  from /home/user/proj/git/my_ws/build/my_msgs/rosidl_typesupport_opensplice_cpp/my_msgs/action/my_action__rosidl_typesupport_opensplice_cpp.hpp:10,
                  from /home/user/proj/git/my_ws/build/my_msgs/rosidl_typesupport_opensplice_cpp/my_msgs/action/dds_opensplice/my_action__type_support.cpp:17:
 /home/user/proj/git/my_ws/build/my_msgs/rosidl_typesupport_opensplice_cpp/my_msgs/action/dds_opensplice/MyAction_.h:16:10: fatal error: UUID_.h: No such file or directory
```

**Required Info:**

- Operating System:
  - Ubunutu 18.04
- Installation type:
  - from source
- Version or commit hash:
  - commit 4d9f442853dfaa7a2480489b0c1bb9285da7ba53 (HEAD -> master, origin/master, origin/HEAD)
- DDS implementation:
  - Opensplice
- Client library (if applicable):
  - rosidl

#### Steps to reproduce issue
Generate an action for Opensplice (fastrtps seems to work):
```
# Goal
int8 state
---
# Result
int8 state
---
# Feedback
int8 opening
```
#### Expected behavior
no error
#### Actual behavior
compile error

#### Additional information
The missing file does exist in the following folder (both build and install):
```
./include/unique_identifier_msgs/msg/dds_opensplice/UUID_.h
```

But this folder is not included in the build command:
```
/usr/bin/c++

-Dmy_msgs__rosidl_typesupport_opensplice_cpp_EXPORTS
-I/home/user/proj/git/my_ws/build/my_msgs/rosidl_generator_cpp
-I/home/user/proj/git/my_ws/build/my_msgs/rosidl_typesupport_opensplice_cpp
-I/home/user/proj/git/my_ws/build/my_msgs/rosidl_typesupport_opensplice_cpp/my_msgs/msg/dds_opensplice
-I/home/user/proj/git/my_ws/build/my_msgs/rosidl_typesupport_opensplice_cpp/my_msgs/srv/dds_opensplice
-I/home/user/proj/git/my_ws/build/my_msgs/rosidl_typesupport_opensplice_cpp/my_msgs/action/dds_opensplice
-I/home/user/proj/ros2_ws/install/include
-I/opt/ADLINK/Vortex_v2/Device/VortexOpenSplice/6.10.1p2/HDE/x86_64.linux/include
-I/opt/ADLINK/Vortex_v2/Device/VortexOpenSplice/6.10.1p2/HDE/x86_64.linux/include/sys
-I/opt/ADLINK/Vortex_v2/Device/VortexOpenSplice/6.10.1p2/HDE/x86_64.linux/include/dcps/C++/SACPP
-I/home/user/proj/ros2_ws/install/include/builtin_interfaces/msg/dds_opensplice

-fPIC

-Wall
-Wextra
-Wpedantic
-std=gnu++14
-o
CMakeFiles/my_msgs__rosidl_typesupport_opensplice_cpp.dir/rosidl_typesupport_opensplice_cpp/my_msgs/action/dds_opensplice/my_action__type_support.cpp.o
-c
/home/user/proj/git/my_ws/build/my_msgs/rosidl_typesupport_opensplice_cpp/my_msgs/action/dds_opensplice/my_action__type_support.cpp
```

scgroot, 2019-03-25:
Seems related to #344 and #348

jacobperron, 2019-03-29:
@scgroot are you able to provide a copy of the CMakeLists.txt that reproduces the error?

jacobperron, 2019-03-29:
Just to be sure, make sure that you have `find_package(action_msgs REQUIRED)` in your `CMakeLists.txt` and a declare a dependency to `action_msgs` when generating interfaces ([for example](https://github.com/ros2/rcl_interfaces/blob/4b9ada89fb4e2c0db6471615fb06f208f19e38dd/test_msgs/CMakeLists.txt#L55)).

sloretz, 2019-03-29:
> Just to be sure, make sure that you have find_package(action_msgs REQUIRED) in your CMakeLists.txt and a declare a dependency to action_msgs when generating interfaces (for example).

I thought that used to be automatic, but it looks like it was just the dependency that was exported. Looks like nothing is found or exported now.

https://github.com/ros2/rosidl/blob/b48b4e9a4c0260fea03d5a1ccd0bc831d79d1ab4/rosidl_cmake/cmake/rosidl_generate_action_interfaces.cmake#L77-L79

scgroot, 2019-04-02:
The CMakeLists.txt
```
cmake_minimum_required(VERSION 3.5)
project(my_msgs)

find_package(ament_cmake REQUIRED)
find_package(action_msgs REQUIRED)
find_package(builtin_interfaces REQUIRED)
find_package(rosidl_default_generators REQUIRED)
find_package(action_msgs REQUIRED)

rosidl_generate_interfaces(my_msgs
  "action/MyAction.action"
  DEPENDENCIES
  	builtin_interfaces
  	action_msgs
)
ament_package()
```

and from the package.xml:
```
  <buildtool_depend>ament_cmake</buildtool_depend>
  <build_depend>builtin_interfaces</build_depend>
  <build_depend>rosidl_default_generators</build_depend>
  <depend>action_msgs</depend>
  <exec_depend>builtin_interfaces</exec_depend>
  <exec_depend>rosidl_default_runtime</exec_depend>
  <member_of_group>rosidl_interface_packages</member_of_group>
  <export>
    <build_type>ament_cmake</build_type>
  </export>
```

scgroot, 2019-09-13:
@jacobperron it indeed seems to compile now