ros2/rosidl #357: [rosidl_generator_c] 'Token' object has no attriubute 'children'

JacobJBeck, 2019-03-29:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source
- Version or commit hash:
  - 4d9f442853dfaa7a2480489b0c1bb9285da7ba53
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
Download and build package.

#### Expected behavior
Package builds without error.

#### Actual behavior
Colcon fails with 

```
Starting >>> rosidl_generator_c
--- stderr: rosidl_generator_c                             
'Token' object has no attribute 'children' /home/jacob/ros2_core_ws/build/rosidl_generator_c/rosidl_adapter/rosidl_generator_c/msg/Bool.idl
Error processing idl file: /home/jacob/ros2_core_ws/build/rosidl_generator_c/rosidl_adapter/rosidl_generator_c/msg/Bool.idl
Traceback (most recent call last):
  File "/home/jacob/ros2_core_ws/src/ros2/rosidl/rosidl_generator_c/bin/rosidl_generator_c", line 40, in <module>
    sys.exit(main())
  File "/home/jacob/ros2_core_ws/src/ros2/rosidl/rosidl_generator_c/bin/rosidl_generator_c", line 35, in main
    args.generator_arguments_file,
  File "/home/jacob/ros2_core_ws/src/ros2/rosidl/rosidl_generator_c/rosidl_generator_c/__init__.py", line 35, in generate_c
    generate_files(generator_arguments_file, mapping)
  File "/home/jacob/ros2_core_ws/install/rosidl_cmake/lib/python3.6/site-packages/rosidl_cmake/__init__.py", line 121, in generate_files
    raise(e)
  File "/home/jacob/ros2_core_ws/install/rosidl_cmake/lib/python3.6/site-packages/rosidl_cmake/__init__.py", line 101, in generate_files
    idl_file = parse_idl_file(locator)
  File "/home/jacob/ros2_core_ws/install/rosidl_parser/lib/python3.6/site-packages/rosidl_parser/parser.py", line 60, in parse_idl_file
    content = parse_idl_string(string, png_file=png_file)
  File "/home/jacob/ros2_core_ws/install/rosidl_parser/lib/python3.6/site-packages/rosidl_parser/parser.py", line 69, in parse_idl_string
    content = extract_content_from_ast(tree)
  File "/home/jacob/ros2_core_ws/install/rosidl_parser/lib/python3.6/site-packages/rosidl_parser/parser.py", line 139, in extract_content_from_ast
    add_message_members(msg, struct_defs[0])
  File "/home/jacob/ros2_core_ws/install/rosidl_parser/lib/python3.6/site-packages/rosidl_parser/parser.py", line 366, in add_message_members
    abstract_type = get_abstract_type_from_type_spec(type_spec)
  File "/home/jacob/ros2_core_ws/install/rosidl_parser/lib/python3.6/site-packages/rosidl_parser/parser.py", line 407, in get_abstract_type_from_type_spec
    return get_abstract_type(child)
  File "/home/jacob/ros2_core_ws/install/rosidl_parser/lib/python3.6/site-packages/rosidl_parser/parser.py", line 416, in get_abstract_type
    while len(child.children) == 1:
AttributeError: 'Token' object has no attribute 'children'
make[2]: *** [rosidl_generator_c/rosidl_generator_c/msg/bool.h] Error 1
make[1]: *** [CMakeFiles/rosidl_generator_c_interfaces__rosidl_generator_c.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< rosidl_generator_c	[ Exited with code 2 ]

Summary: 0 packages finished [1.93s]
  1 package failed: rosidl_generator_c
  1 package had stderr output: rosidl_generator_c
```

#### Additional information
None.

dirk-thomas, 2019-03-29:
What version of the Debian package `python3-lark-parser` do you have installed?

JacobJBeck, 2019-03-29:
Doesn't seem to be installed as a Debian package.
```
apt-cache show python3-lark-parser
N: Unable to locate package python3-lark-parser
E: No packages found
```
But python does find it.
```
python3
Python 3.6.7 (default, Oct 22 2018, 11:32:17) 
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import lark
>>> lark.__version__
'0.6.5'
>>> 
```

dirk-thomas, 2019-03-29:
Then you need to uninstall the Python package first (`pip3 uninstall lark-parser`) and then install the Debian package (should be version 0.7) as mentioned in the install instructions.

closed 2019-03-29 by dirk-thomas