ros2/rosidl #358: Cpp code generator uses message field_name as variable

mwamp, 2019-04-12:
## Bug report

**Required Info:**

- Operating System:
  - Bionic
- Installation type:
  - binaries
- Version or commit hash:
  - Ubuntu repo : 0.6.3

#### Issue

Given a service .srv file (or probably similarly a .msg file)
``` 
string name
string data
---
string value
```
it appears to me that the cpp code generator uses the message field_names as cpp class variables. In short the c++ generate will have something like : 
```
class MyFancyServiceArguments{
    std::string name;
    std::string data;
}
class MyFancyServiceOutputs{
    std::string value;
}
...
instanceOf__MyFancyServiceOutputs.data = ...
```

While the example above is fine, the following example won't compile
``` 
string name
string default
---
string value
```
This is because ```default``` is reserved in c++. I did not test it but other c++ keywords such as ```class, double``` etc would also be problematic.
And it would not be so strange for people to actually use these as field_names so I guess it is a problem.

closed 2019-04-12 by mwamp

dirk-thomas, 2019-04-12:
Duplicate of #272.

mwamp, 2019-04-12:
Yes sorry I just saw that.
God you're quick