ros2/rosidl #365: getter functions of rosidl_typesupport_introspection_cpp::MessageMember are NULL

Karsten1987, 2019-04-24:
Getter functions of `rosidl_typesupport_introspection_cpp::MessageMember` are set to `NULL` and thus segfault.

output of lldb:
```
(rosidl_typesupport_introspection_cpp::MessageMember) $0 = {
  name_ = 0x00000001008a34de "bool_values"
  type_id_ = '\x06'
  string_upper_bound_ = 0
  members_ = 0x0000000000000000
  is_array_ = true
  array_size_ = 3
  is_upper_bound_ = false
  offset_ = 0
  default_value_ = 0x0000000000000000
  size_function = 0x0000000000000000
  get_const_function = 0x0000000000000000
  get_function = 0x0000000000000000
  resize_function = 0x0000000000000000
```
See here: https://github.com/ros2/rosidl/blob/master/rosidl_typesupport_introspection_cpp/include/rosidl_typesupport_introspection_cpp/message_introspection.hpp#L39-L42

We are using these functions within rosbag2 to allocate dynamic messages:
https://github.com/ros2/rosbag2/blob/compile_tests/rosbag2/src/rosbag2/types/introspection_message.cpp#L258-L275

Karsten1987, 2019-04-24:
This was earlier addressed here: https://github.com/ros2/rosidl/pull/328
But I believe this behavior might have changed.

dirk-thomas, 2019-04-24:
The introspection typesupport intentionally doesn't provide such functions since you can do the very same with the provided information already: the type information for each field and their offset address.

Karsten1987, 2019-04-24:
see the referenced pull request where we tackled exactly this.

closed 2019-04-25 by Karsten1987