ros2/rosidl #378: resize_function is NULL for ParameterEvent new_parameters, changed_parameters, deleted_parameters

malsbat, 2019-05-13:
This causes the rmw_dps implementation to crash.  This worked in Crystal.

In the master branch, build/rcl_interfaces/rosidl_typesupport_introspection_cpp/rcl_interfaces/msg/parameter_event__type_support.cpp, resize_function__ParameterEvent__new_parameters is defined, but it is not included in the  ParameterEvent_message_member_array structure.

In release-crystal-20190408 the resize_function is included.

I see that the check to define the function uses isinstance(member.type, AbstractSequence) whereas the check to include it is isinstance(member.type.value_type, AbstractSequence).  I'm not sure which one is correct.

jacobperron, 2019-05-14:
Looking at the following diff, I would say the check to include it should be `isinstance(member.type, AbstractSequence)`:

https://github.com/ros2/rosidl/commit/416ebd84a678178f8e8941a1961af8abe7514251/rosidl_typesupport_introspection_cpp/resource/msg__type_support.cpp.em#diff-aba1f884b18aaf4e81588ad9968b55bfR175

See if #379 works.

alsora, 2019-05-14:
I tested the PR and it looks that it fixes the problem!

malsbat, 2019-05-14:
The PR looks good to me, thanks.

PR #379 (2019-05-14): Fix condition for including resize functions
Resolves #378 

Bug introduced by #334.
closed 2019-05-14 by jacobperron