ros2/rosidl #386: Cannot create service with same array member in both request and response

jdlangs, 2019-06-19:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Dashing from binaries
- Version or commit hash:
  - `ros-dashing-rosidl-typesupport-introspection-c     0.7.4-1bionic.20190612.174026`
- DDS implementation:
  - FastRTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Define a custom service type that has a member in the request and response with the same field name and non-primitive array type.

E.g.,
```
geometry_msgs/Point[] points
---
geometry_msgs/Point[] points
```

#### Expected behavior
Successful creation of the custom service type

#### Actual behavior
Failure to compile the generated code:
```
/home/jlangsfeld/Workspaces/ros2/build/srv_pkg/rosidl_typesupport_introspection_c/srv_pkg/srv/sample_service__type_support.c:136:8: error: redefinition of ‘srv_pkg__srv__sample_service__rosidl_typesupport_introspection_c__size_function__Point__points’
 size_t srv_pkg__srv__sample_service__rosidl_typesupport_introspection_c__size_function__Point__points(
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jlangsfeld/Workspaces/ros2/build/srv_pkg/rosidl_typesupport_introspection_c/srv_pkg/srv/sample_service__type_support.c:25:8: note: previous definition of ‘srv_pkg__srv__sample_service__rosidl_typesupport_introspection_c__size_function__Point__points’ was here
 size_t srv_pkg__srv__sample_service__rosidl_typesupport_introspection_c__size_function__Point__points(
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jlangsfeld/Workspaces/ros2/build/srv_pkg/rosidl_typesupport_introspection_c/srv_pkg/srv/sample_service__type_support.c:144:14: error: redefinition of ‘srv_pkg__srv__sample_service__rosidl_typesupport_introspection_c__get_const_function__Point__points’
 const void * srv_pkg__srv__sample_service__rosidl_typesupport_introspection_c__get_const_function__Point__points(
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jlangsfeld/Workspaces/ros2/build/srv_pkg/rosidl_typesupport_introspection_c/srv_pkg/srv/sample_service__type_support.c:33:14: note: previous definition of ‘srv_pkg__srv__sample_service__rosidl_typesupport_introspection_c__get_const_function__Point__points’ was here
 const void * srv_pkg__srv__sample_service__rosidl_typesupport_introspection_c__get_const_function__Point__points(
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jlangsfeld/Workspaces/ros2/build/srv_pkg/rosidl_typesupport_introspection_c/srv_pkg/srv/sample_service__type_support.c:152:8: error: redefinition of ‘srv_pkg__srv__sample_service__rosidl_typesupport_introspection_c__get_function__Point__points’
 void * srv_pkg__srv__sample_service__rosidl_typesupport_introspection_c__get_function__Point__points(
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jlangsfeld/Workspaces/ros2/build/srv_pkg/rosidl_typesupport_introspection_c/srv_pkg/srv/sample_service__type_support.c:41:8: note: previous definition of ‘srv_pkg__srv__sample_service__rosidl_typesupport_introspection_c__get_function__Point__points’ was here
 void * srv_pkg__srv__sample_service__rosidl_typesupport_introspection_c__get_function__Point__points(
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jlangsfeld/Workspaces/ros2/build/srv_pkg/rosidl_typesupport_introspection_c/srv_pkg/srv/sample_service__type_support.c:160:6: error: redefinition of ‘srv_pkg__srv__sample_service__rosidl_typesupport_introspection_c__resize_function__Point__points’
 bool srv_pkg__srv__sample_service__rosidl_typesupport_introspection_c__resize_function__Point__points(
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jlangsfeld/Workspaces/ros2/build/srv_pkg/rosidl_typesupport_introspection_c/srv_pkg/srv/sample_service__type_support.c:49:6: note: previous definition of ‘srv_pkg__srv__sample_service__rosidl_typesupport_introspection_c__resize_function__Point__points’ was here
 bool srv_pkg__srv__sample_service__rosidl_typesupport_introspection_c__resize_function__Point__points(
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

pbeeson, 2019-07-11:
I can verify this issue is a problem .  This makes large-scale migration from ROS1 much more of a hassle.

jacobperron, 2019-07-12:
I can also reproduce with the latest source. More details:

- The issue occurs for arrays and sequences.
- The issue only occurs with non-basic types (e.g. not `int32[]` or `char[]`).
- The same issue happens with actions; if two or more parts of the action have the same field.

closed 2019-07-20 by jacobperron