ros2/rosidl #388: Error: 'charmap' codec can't decode byte 0x8f in position 156

minggangw, 2019-07-15:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10 ONLY
- Installation type:
  - Binary of nightly build (http://ci.ros2.org/view/packaging/job/packaging_windows/lastSuccessfulBuild/artifact/ws/ros2-package-windows-AMD64.zip)
- Version or commit hash:
- DDS implementation:
- Client library (if applicable):

#### Steps to reproduce issue

Pass the path of [WStrings.msg](https://github.com/ros2/test_interface_files/blob/master/msg/WStrings.msg) as the parameter of Python method [parse_message_file](https://github.com/ros2/rosidl/blob/ced307012bda3413d6c495914bf517d0d03748f1/rosidl_adapter/rosidl_adapter/parser.py#L427)

#### Expected behavior
Execute successfully
#### Actual behavior
Error: 'charmap' codec can't decode byte 0x8f in position 156: character maps to <undefined>
#### Additional information
- Python version: Python3.7 x64
- Suspection: there are some special characters though commented out, and I found possible some potential reason, see https://stackoverflow.com/questions/9233027/unicodedecodeerror-charmap-codec-cant-decode-byte-x-in-position-y-character

Thanks!

dirk-thomas, 2019-07-18:
I misread the description and comments and assumed the problem was in generated C / C++ code.

Since this actually happens when reading the interface file I think #390 should be sufficient to resolve your problem. @minggangw please comment here if it addressed your problem.

dirk-thomas, 2019-07-19:
Reopening since there are follow up issues when using non-ASCII character in e.g. default values:
* ros2/test_interface_files#8 adds a message with non-ASCII characters in the default values

dirk-thomas, 2019-07-26:
With all PRs rebased to address other unrelated test failures I triggered an full round of CI builds to double check for regressions:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

minggangw, 2019-07-30:
This problem is fixed, I verified it on https://ci.appveyor.com/project/minggangw/rclnodejs/builds/26335124, thanks!

dirk-thomas, 2019-07-30:
Until the list of pending PRs is merged we will keep this ticket open.

dirk-thomas, 2019-07-30:
I disabled one case which doesn't seem to work cross vendor: https://github.com/ros2/system_tests/pull/377/commits/5d100b64178f679cd436772ca2258532d7389126

Hopefully the final round of CI:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

PR #390 (2019-07-19): open interface files with utf-8 encoding
Fixes #388.

[image]
closed 2019-07-31 by dirk-thomas