ros2/rosidl #394: build msgs failed in ros2

jhx2002, 2019-07-23:
- ros2 version : dash

- build  for   https://github.com/ros2/demos/tree/master/pendulum_msgs

- build command:  colcon build

- error log is following: 

```
Traceback (most recent call last):
  File "/opt/ros/dashing/lib/rosidl_generator_dds_idl/rosidl_generator_dds_idl", line 41, in <module>
    sys.exit(main())
  File "/opt/ros/dashing/lib/rosidl_generator_dds_idl/rosidl_generator_dds_idl", line 36, in main
    args.additional_service_templates,
  File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_generator_dds_idl/__init__.py", line 51, in generate_dds_idl
    generate_files(generator_arguments_file, mapping, additional_context, keep_case=True)
  File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_cmake/__init__.py", line 90, in generate_files
    raise(e)
  File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_cmake/__init__.py", line 70, in generate_files
make[2]: *** [rosidl_generator_c/pendulum_msgs/msg/joint_state.h] Error 1
    idl_file = parse_idl_file(locator)
  File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_parser/parser.py", line 64, in parse_idl_file
    content = parse_idl_string(string, png_file=png_file)
  File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_parser/parser.py", line 73, in parse_idl_string
make[1]: *** [CMakeFiles/pendulum_msgs__rosidl_generator_c.dir/all] Error 2
    content = extract_content_from_ast(tree)
  File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_parser/parser.py", line 144, in extract_content_from_ast
    add_message_members(msg, struct_defs[0])
  File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_parser/parser.py", line 365, in add_message_members
    abstract_type = get_abstract_type_from_type_spec(type_spec)
  File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_parser/parser.py", line 406, in get_abstract_type_from_type_spec
    return get_abstract_type(child)
  File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_parser/parser.py", line 415, in get_abstract_type
    while len(child.children) == 1:
AttributeError: 'Token' object has no attribute 'children'
make[2]: *** [rosidl_generator_dds_idl/pendulum_msgs/msg/dds_opensplice/JointState_.idl] Error 1
make[1]: *** [CMakeFiles/pendulum_msgs__dds_opensplice_idl.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< pendulum_msgs	[ Exited with code 2 ]

```

dirk-thomas, 2019-07-23:
Please check the version of your `python3-lark-parser` Debian package to ensure it is version 0.7 and double check that you have now older version installed via `pip` (`python3 -c "import lark; print(lark.__version__)"`).

jhx2002, 2019-07-24:
```
root@localhost:/home/huaxing.jinhx/oproj/demos-0.7.8/pendulum_msgs# pip3 install lark-parser
Requirement already satisfied: lark-parser in /usr/local/lib/python3.6/dist-packages
root@localhost:/home/huaxing.jinhx/oproj/demos-0.7.8/pendulum_msgs# python3 -c "import lark; print(lark.__version__)"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
AttributeError: module 'lark' has no attribute '__version__'

```

dirk-thomas, 2019-07-24:
* You should **not** use `pip` to install Python packages which are also available via `apt`. You should uninstall the `pip` package and install it via `apt`: `python3-lark-parser`.
* You might have another Python module installed with the same name. Please check `python3 -c "import lark; print(lark.__file__)"`/ `pip3 freeze`.

jhx2002, 2019-07-25:
Thank you!  Issue may close

closed 2019-07-25 by jhx2002