ros2/rosidl #400: Using `rosidl_target_interfaces` incompatible with keyword `target_link_libraries` command

jdlangs, 2019-08-16:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Binaries
- Version or commit hash:
  - ros-dashing-rosidl-cmake 0.7.5-1bionic.20190801.140923
- DDS implementation:
  - FastRTPS

Similar to the problem I previous encountered with [`ament_target_dependencies`](https://github.com/ament/ament_cmake/issues/158), the `rosidl_target_interfaces` macro calls `target_link_libraries` with the non-keyword format, which is incompatible with additional calls that do use the keyword format on the same target.

That issue was resolved by [adding an optional keyword form](https://github.com/ament/ament_cmake/pull/161) to `ament_target_dependencies`. Could the same thing be done here for `rosidl_target_interfaces`?

ivanpauno, 2019-08-16:
> That issue was resolved by adding an optional keyword form to ament_target_dependencies. Could the same thing be done here for rosidl_target_interfaces?

Sounds reasonable. Are you planning to open a PR with the fix?

jdlangs, 2019-08-19:
I'll aim to put an initial attempt together sometime this week. If anyone else sees this though and there's been no activity, you can safely assume I haven't started working on it yet and you're welcome to take a stab at it.

dirk-thomas, 2019-10-23:
@jdlangs Any update on this?

prajd3d, 2021-04-26:
I just hit this issue. And, if anyone needed a immediate workaround, here is what I did.
A temporary work around is to move the messages to a separate package (e.g., my_msgs).
And, reference my_msgs in the current package.

estellearrc, 2022-07-05:
Any update for this issue?

ivanpauno, 2022-07-05:
> Any update for this issue?

The issue is tagged as `help wanted`, i.e. it's open to contributions.
You would need to implement something similar to https://github.com/ament/ament_cmake/pull/161.

EricCousineau-TRI, 2022-10-18:
As an alternative, why not declare a Modern CMake target that would make `rosidl_target_interfaces` moot? e.g.
```
target_link_libraries(my_thing PUBLIC
   geometry_msgs::targets
)
```
In things like https://github.com/RobotLocomotion/drake-ros/pull/153, I see thus usage of `${geometry_msgs_TARGETS}`.

I believe this may connect w/ @sloretz's points in https://github.com/ament/ament_cmake/issues/292

clalancette, 2022-10-18:
> As an alternative, why not declare a Modern CMake target that would make `rosidl_target_interfaces` moot?

That's the eventual goal.  I swear I remember us trying and failing to do this for Humble (as part of https://github.com/ros2/ros2/issues/1150), but I can't find the relevant PRs right now.  Maybe @sloretz remembers working on this and can find the relevant PRs.

EricCousineau-TRI, 2022-10-18:
Ah, gotcha. Yeah, knowing the correct modern target would be great!