ros2/rosidl #404: [Feature Request] Allocate generic typesupport_introspection_cpp message

Karsten1987, 2019-08-29:
There is currently code available within the rosbag packages, which allows to [allocate memory for a ROS2 message given the introspection typesupport](https://github.com/ros2/rosbag2/blob/master/rosbag2/src/rosbag2/types/introspection_message.cpp#L29-L47).

This is essentially nothing else than iterating over the typesupport and respectively allocate individually fields. This essentially allows to allocate a ROS2 message with the same memory layout as a regularly created ROS2 message without requiring a compile time available template.

While this was working fine for rosbags, we realized that this is being used in other rmw implementations as well and I am wondering if this repo is a suited place to move this piece of code into.

dirk-thomas, 2019-08-29:
> we realized that this is being used in other rmw implementations as well 

Can you please reference these for context.

Karsten1987, 2019-08-29:
one of them being [fastrtps_dynamic](https://github.com/ros2/rmw_fastrtps/blob/master/rmw_fastrtps_dynamic_cpp/include/rmw_fastrtps_dynamic_cpp/TypeSupport_impl.hpp)

dirk-thomas, 2019-08-29:
The referenced function doesn't seem to allocate a message but only populate a given message pointer with data?

closed 2019-10-18 by Karsten1987