ros2/rosidl #413: Change in field type of custom msg results in compiling error

Leo-Nogismus, 2019-09-13:
## Bug report

**Required Info:**

- Operating System:
  -Ubuntu 16.04
- Installation type:
  - from Source
- Version or commit hash:
  - crystal
- DDS implementation:
  - RTI Connext
- Client library (if applicable):
  -rclcpp

#### Steps to reproduce issue
Custom msg pkg in workspace, generated srv file. If value in response of srv is changed from string to int or float or the other way around it results in a compiling Error (see below). Have not tried it with custom types, yet. Here are my CMakeList, package.xml and the srv file:
srv-file (ModelToInternal.srv) before change:
```
---
std_msgs/Header header
float32 data
```
srv-file (ModelToInternal.srv) after change:
```
---
std_msgs/Header header
string data
```
CMakeLists.txt:
```
cmake_minimum_required(VERSION 3.5)
project(custom_msgs)

# Default to C99
if(NOT CMAKE_C_STANDARD)
  set(CMAKE_C_STANDARD 99)
endif()

# Default to C++14
if(NOT CMAKE_CXX_STANDARD)
  set(CMAKE_CXX_STANDARD 14)
endif()

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

# find dependencies
find_package(ament_cmake REQUIRED)
find_package(std_msgs REQUIRED)
find_package(rclcpp REQUIRED)
find_package(rclpy REQUIRED)
find_package(rosidl_default_generators REQUIRED)

set(srv_files "srv/ModelToInternal.srv")

rosidl_generate_interfaces(${PROJECT_NAME}
  ${srv_files}
  DEPENDENCIES
  std_msgs
)

if(BUILD_TESTING)
  find_package(ament_lint_auto REQUIRED)
  # the following line skips the linter which checks for copyrights
  # uncomment the line when a copyright and license is not present in all source files
  #set(ament_cmake_copyright_FOUND TRUE)
  # the following line skips cpplint (only works in a git repo)
  # uncomment the line when this package is not in a git repo
  #set(ament_cmake_cpplint_FOUND TRUE)
  ament_lint_auto_find_test_dependencies()
endif()

ament_package()

```
package.xml:
```
<?xml version="1.0"?>
<?xml-model href="http://download.ros.org/schema/package_format2.xsd" schematypens="http://www.w3.org/2001/XMLSchema"?>
<package format="3">
  <name>custom_msgs</name>
  <version>0.0.0</version>
  <description>TODO: Package description</description>
  <maintainer email="example@example.de">leonard</maintainer>
  <license>TODO: License declaration</license>

  <buildtool_depend>ament_cmake</buildtool_depend>

  <build_depend>rosidl_default_generators</build_depend>

  <build_depend>std_msgs</build_depend>
  <build_depend>rclcpp</build_depend>
  <build_depend>rclpy</build_depend>

  <member_of_group>rosidl_interface_packages</member_of_group>

  <exec_depend>std_msgs</exec_depend>
  <exec_depend>rclcpp</exec_depend>
  <exec_depend>rclpy</exec_depend>

  <exec_depend>rosidl_default_runtime</exec_depend>

  <test_depend>ament_lint_auto</test_depend>
  <test_depend>ament_lint_common</test_depend>

  <export>
    <build_type>ament_cmake</build_type>
  </export>
</package>
```

#### Expected behavior
One should be able to change the data types of custom msgs without needing to delete the build folder. ```colcon build --cmake-target clean``` does not help, either.

#### Actual behavior
```
colcon build
Starting >>> custom_msgs
[0.475s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/leonard/masterarbeit_ros2_ws/install/controller_node_test' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.484s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/leonard/masterarbeit_ros2_ws/install/controller_node_test' in the environment variable CMAKE_PREFIX_PATH doesn't exist
Starting >>> clock_node_test
Finished <<< clock_node_test [0.42s]                                                            
[Processing: custom_msgs]                             
--- stderr: custom_msgs                               
/home/leonard/masterarbeit_ros2_ws/build/custom_msgs/rosidl_generator_py/custom_msgs/srv/_model_to_internal__response_s.c: In function ‘custom_msgs__srv__model_to_internal__response__convert_from_py’:
/home/leonard/masterarbeit_ros2_ws/build/custom_msgs/rosidl_generator_py/custom_msgs/srv/_model_to_internal__response_s.c:69:23: error: incompatible types when assigning to type ‘rosidl_generator_c__String {aka struct rosidl_generator_c__String}’ from type ‘double’
     ros_message->data = PyFloat_AS_DOUBLE(field);
                       ^
/home/leonard/masterarbeit_ros2_ws/build/custom_msgs/rosidl_generator_py/custom_msgs/srv/_model_to_internal__response_s.c: In function ‘custom_msgs__srv__model_to_internal__response__convert_to_py’:
/home/leonard/masterarbeit_ros2_ws/build/custom_msgs/rosidl_generator_py/custom_msgs/srv/_model_to_internal__response_s.c:110:32: error: incompatible type for argument 1 of ‘PyFloat_FromDouble’
     field = PyFloat_FromDouble(ros_message->data);
                                ^~~~~~~~~~~
In file included from /usr/include/python3.5m/Python.h:81:0,
                 from /home/leonard/masterarbeit_ros2_ws/build/custom_msgs/rosidl_generator_py/custom_msgs/srv/_model_to_internal__response_s.c:4:
/usr/include/python3.5m/floatobject.h:45:24: note: expected ‘double’ but argument is of type ‘rosidl_generator_c__String {aka struct rosidl_generator_c__String}’
 PyAPI_FUNC(PyObject *) PyFloat_FromDouble(double);
                        ^~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/custom_msgs__python.dir/rosidl_generator_py/custom_msgs/srv/_model_to_internal__response_s.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/custom_msgs__python.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< custom_msgs	[ Exited with code 2 ]

Summary: 1 package finished [32.5s]
  1 package failed: custom_msgs
  1 package had stderr output: custom_msgs
  1 package not processed
```

#### Additional information

The other way around (change from string to float32) does not work either:
```
colcon build
Starting >>> custom_msgs
[0.466s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/leonard/masterarbeit_ros2_ws/install/controller_node_test' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.475s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/leonard/masterarbeit_ros2_ws/install/controller_node_test' in the environment variable CMAKE_PREFIX_PATH doesn't exist
Starting >>> clock_node_test
Finished <<< clock_node_test [0.44s]                                                            
[Processing: custom_msgs]                             
--- stderr: custom_msgs                               
/home/leonard/masterarbeit_ros2_ws/build/custom_msgs/rosidl_generator_py/custom_msgs/srv/_model_to_internal__response_s.c: In function ‘custom_msgs__srv__model_to_internal__response__convert_from_py’:
/home/leonard/masterarbeit_ros2_ws/build/custom_msgs/rosidl_generator_py/custom_msgs/srv/_model_to_internal__response_s.c:77:40: warning: passing argument 1 of ‘rosidl_generator_c__String__assign’ from incompatible pointer type [-Wincompatible-pointer-types]
     rosidl_generator_c__String__assign(&ros_message->data, PyBytes_AS_STRING(encoded_field));
                                        ^
In file included from /home/leonard/masterarbeit_ros2_ws/build/custom_msgs/rosidl_generator_py/custom_msgs/srv/_model_to_internal__response_s.c:13:0:
/home/leonard/ros2/install/rosidl_generator_c/include/rosidl_generator_c/string_functions.h:50:1: note: expected ‘rosidl_generator_c__String * {aka struct rosidl_generator_c__String *}’ but argument is of type ‘float *’
 rosidl_generator_c__String__assign(
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/leonard/masterarbeit_ros2_ws/build/custom_msgs/rosidl_generator_py/custom_msgs/srv/_model_to_internal__response_s.c: In function ‘custom_msgs__srv__model_to_internal__response__convert_to_py’:
/home/leonard/masterarbeit_ros2_ws/build/custom_msgs/rosidl_generator_py/custom_msgs/srv/_model_to_internal__response_s.c:120:24: error: request for member ‘data’ in something not a structure or union
       ros_message->data.data,
                        ^
/home/leonard/masterarbeit_ros2_ws/build/custom_msgs/rosidl_generator_py/custom_msgs/srv/_model_to_internal__response_s.c:121:31: error: request for member ‘data’ in something not a structure or union
       strlen(ros_message->data.data),
                               ^
make[2]: *** [CMakeFiles/custom_msgs__python.dir/rosidl_generator_py/custom_msgs/srv/_model_to_internal__response_s.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/custom_msgs__python.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< custom_msgs	[ Exited with code 2 ]

Summary: 1 package finished [32.8s]
  1 package failed: custom_msgs
  1 package had stderr output: custom_msgs
  1 package not processed
```

dirk-thomas, 2019-09-13:
Can you please try to reproduce the problem with a newer ROS 2 release (either Dashing or master) to confirm this is still a problem?

Leo-Nogismus, 2019-09-16:
Okay, no problem on dashing. Sorry. Thanks for the help!

closed 2019-09-16 by Leo-Nogismus