ros2/rosidl #414: rosidl_adapter crashes if .msg file begins with lower case letter

TimCraig, 2019-09-15:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior
If the filename capitalization is an actual requirement, I would expect the parser to generate a reasonable error message and possibly point to the document defining  the requirement.

#### Actual behavior
The parser Python program crashes with error messages that are not helpful the an end user.

Starting >>> druai_msg
--- stderr: druai_msg                         
CMake Error at /opt/ros/dashing/share/rosidl_adapter/cmake/rosidl_adapt_interfaces.cmake:60 (message):
  execute_process(/usr/bin/python3 -m rosidl_adapter --package-name druai_msg
  --arguments-file
  /home/brutusdev/ros2/test_ws/build/druai_msg/rosidl_adapter__arguments__druai_msg.json
  --output-dir
  /home/brutusdev/ros2/test_ws/build/druai_msg/rosidl_adapter/druai_msg
  --output-file
  /home/brutusdev/ros2/test_ws/build/druai_msg/rosidl_adapter/druai_msg.idls)
  returned error code 1:

  Traceback (most recent call last):

    File "/usr/lib/python3.6/runpy.py", line 193, in _run_module_as_main
      "__main__", mod_spec)
    File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
      exec(code, run_globals)
    File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_adapter/__main__.py", line 19, in <module>
      sys.exit(main())
    File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_adapter/main.py", line 55, in main
      pathlib.Path(relative_path), output_dir)
    File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_adapter/__init__.py", line 20, in convert_to_idl
      package_dir, package_name, interface_file, output_dir / 'msg')
    File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_adapter/msg/__init__.py", line 28, in convert_msg_to_idl
      msg = parse_message_string(package_name, input_file.stem, content)
    File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_adapter/parser.py", line 522, in parse_message_string
      msg = MessageSpecification(pkg_name, msg_name, fields, constants)
    File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_adapter/parser.py", line 373, in __init__
      pkg_name + PACKAGE_NAME_MESSAGE_TYPE_SEPARATOR + msg_name)
    File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_adapter/parser.py", line 196, in __init__
      raise InvalidResourceName(self.type)

  rosidl_adapter.parser.InvalidResourceName: druai

Call Stack (most recent call first):
  /opt/ros/dashing/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:130 (rosidl_adapt_interfaces)
  CMakeLists.txt:24 (rosidl_generate_interfaces)

#### Additional information

jacobperron, 2019-09-16:
In [this design article](http://design.ros2.org/articles/legacy_interface_definition.html#conventions), it is described that interface definitions have upper-case CamelCase, by convention. Perhaps we should update the article so that it is a **requirement**. Though, I'm not sure why it is a requirement (maybe for IDL compatibility). We could uncomment the following line to follow ROS 1 naming patterns:

https://github.com/ros2/rosidl/blob/07a2ab5c452c462ae067769b752a1811cfb3612e/rosidl_adapter/rosidl_adapter/parser.py#L66

But, for now, I've opened a PR to improve the error message (#415)

jacobperron, 2019-09-26:
@ros2/team What is the ultimate goal? Do we want to maintain compatibility with ROS 1 interfaces that can have *camelCase*, or strictly only allow *CamelCase*?

dirk-thomas, 2019-09-26:
> In this design article, it is described that interface definitions have upper-case CamelCase, by convention. Perhaps we should update the article so that it is a requirement.

The current design article already uses **must**: `Both file names must use an upper camel case name and only consist of alphanumeric characters.`

The rational why that was chosen is that all names (the msg filename, the generated source filename, as well as the class name) are the same.

If we want to drop that requirement we will have to accept potential collisions since two entities with different case can exist.

jacobperron, 2019-09-26:
> The current design article already uses must: Both file names must use an upper camel case name and only consist of alphanumeric characters.

I was confused because the title of the section is "Conventions". Thanks for the response!

I'll close this issue then since the error is by design.

closed 2019-09-26 by jacobperron