ros2/rosidl #418: rosidl 0.8.0 release to Dashing

TSC21, 2019-10-14:
@dirk-thomas @mjcarroll it was reported in #406, and fixed in #409 and #410, the bug with string literals and multiline comments in IDL files. The bug was reported on ROS2 Dashing, which is an LTS version that I supposed it should be used to stabilise package APIs using Dashing (which is something I am looking to do).

I see that rosidl package released under 0.8.0 tag are only available under ROS2 Eloquent, and only version 0.7.6 is available to Dashing, which doesn't fix the bug on regarding the multiline comments (string literals).

Is it possible to make the 0.8.0 also available under Dashing, or at least back port the above fixes, in order to address the same bug? Doesn't seem reasonable to bump into a new ROS version in order to solve a bug.

Thanks in advance!

dirk-thomas, 2019-10-15:
Thanks for the report.

#409 was already backported to Dashing in #417.

#410 (and #411) are being backported to Dashing by #403

closed 2019-10-15 by dirk-thomas

TSC21, 2019-10-15:
Thank you @dirk-thomas!