ros2/rosidl #421: rosidl generates redundant declarations

RoboticsYY, 2019-11-12:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - binaries
- Version or commit hash:
  - Dashing release
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue

```
source /opt/ros/dashing/setup.bash

export CXXFLAGS="-Wredundant-decls"

mkdir -p ws_ros2/src && cd ws_ros2/src
git clone https://github.com/ros2/example_interfaces.git
cd .. && colcon build 
```
#### Actual behavior

**rosidl** generates multiple redundant declarations.

```shell
Starting >>> example_interfaces
--- stderr: example_interfaces                                
/home/yy/ws_test/build/example_interfaces/rosidl_typesupport_fastrtps_cpp/example_interfaces/action/dds_fastrtps/fibonacci__type_support.cpp:622:6: warning: redundant redeclaration of ‘bool example_interfaces::action::typesupport_fastrtps_cpp::cdr_serialize(const Fibonacci_Goal&, eprosima::fastcdr::Cdr&)’ in same scope [-Wredundant-decls]
 bool cdr_serialize(
      ^~~~~~~~~~~~~
/home/yy/ws_test/build/example_interfaces/rosidl_typesupport_fastrtps_cpp/example_interfaces/action/dds_fastrtps/fibonacci__type_support.cpp:31:1: note: previous declaration of ‘bool example_interfaces::action::typesupport_fastrtps_cpp::cdr_serialize(const Fibonacci_Goal&, eprosima::fastcdr::Cdr&)’
 cdr_serialize(
 ^~~~~~~~~~~~~
/home/yy/ws_test/build/example_interfaces/rosidl_typesupport_fastrtps_cpp/example_interfaces/action/dds_fastrtps/fibonacci__type_support.cpp:625:6: warning: redundant redeclaration of ‘bool example_interfaces::action::typesupport_fastrtps_cpp::cdr_deserialize(eprosima::fastcdr::Cdr&, example_interfaces::action::Fibonacci_Goal&)’ in same scope [-Wredundant-decls]
 bool cdr_deserialize(
      ^~~~~~~~~~~~~~~
/home/yy/ws_test/build/example_interfaces/rosidl_typesupport_fastrtps_cpp/example_interfaces/action/dds_fastrtps/fibonacci__type_support.cpp:42:1: note: previous declaration of ‘bool example_interfaces::action::typesupport_fastrtps_cpp::cdr_deserialize(eprosima::fastcdr::Cdr&, example_interfaces::action::Fibonacci_Goal&)’
 cdr_deserialize(
 ^~~~~~~~~~~~~~~
/home/yy/ws_test/build/example_interfaces/rosidl_typesupport_fastrtps_cpp/example_interfaces/action/dds_fastrtps/fibonacci__type_support.cpp:628:8: warning: redundant redeclaration of ‘size_t example_interfaces::action::typesupport_fastrtps_cpp::get_serialized_size(const Fibonacci_Goal&, size_t)’ in same scope [-Wredundant-decls]
 size_t get_serialized_size(
        ^~~~~~~~~~~~~~~~~~~
/home/yy/ws_test/build/example_interfaces/rosidl_typesupport_fastrtps_cpp/example_interfaces/action/dds_fastrtps/fibonacci__type_support.cpp:54:1: note: previous declaration of ‘size_t example_interfaces::action::typesupport_fastrtps_cpp::get_serialized_size(const Fibonacci_Goal&, size_t)’
 get_serialized_size(
 ^~~~~~~~~~~~~~~~~~~
/home/yy/ws_test/build/example_interfaces/rosidl_typesupport_fastrtps_cpp/example_interfaces/action/dds_fastrtps/fibonacci__type_support.cpp:632:1: warning: redundant redeclaration of ‘size_t example_interfaces::action::typesupport_fastrtps_cpp::max_serialized_size_Fibonacci_Goal(bool&, size_t)’ in same scope [-Wredundant-decls]
 max_serialized_size_Fibonacci_Goal(
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/yy/ws_test/build/example_interfaces/rosidl_typesupport_fastrtps_cpp/example_interfaces/action/dds_fastrtps/fibonacci__type_support.cpp:77:1: note: previous declaration of ‘size_t example_interfaces::action::typesupport_fastrtps_cpp::max_serialized_size_Fibonacci_Goal(bool&, size_t)’
 max_serialized_size_Fibonacci_Goal(
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/yy/ws_test/build/example_interfaces/rosidl_typesupport_fastrtps_cpp/example_interfaces/action/dds_fastrtps/fibonacci__type_support.cpp:1157:6: warning: redundant redeclaration of ‘bool unique_identifier_msgs::msg::typesupport_fastrtps_cpp::cdr_serialize(const UUID&, eprosima::fastcdr::Cdr&)’ in same scope [-Wredundant-decls]
 bool cdr_serialize(
      ^~~~~~~~~~~~~
/home/yy/ws_test/build/example_interfaces/rosidl_typesupport_fastrtps_cpp/example_interfaces/action/dds_fastrtps/fibonacci__type_support.cpp:599:6: note: previous declaration of ‘bool unique_identifier_msgs::msg::typesupport_fastrtps_cpp::cdr_serialize(const UUID&, eprosima::fastcdr::Cdr&)’
 bool cdr_serialize(
      ^~~~~~~~~~~~~
/home/yy/ws_test/build/example_interfaces/rosidl_typesupport_fastrtps_cpp/example_interfaces/action/dds_fastrtps/fibonacci__type_support.cpp:1160:6: warning: redundant redeclaration of ‘bool unique_identifier_msgs::msg::typesupport_fastrtps_cpp::cdr_deserialize(eprosima::fastcdr::Cdr&, unique_identifier_msgs::msg::UUID&)’ in same scope [-Wredundant-decls]
 bool cdr_deserialize(
      ^~~~~~~~~~~~~~~
/home/yy/ws_test/build/example_interfaces/rosidl_typesupport_fastrtps_cpp/example_interfaces/action/dds_fastrtps/fibonacci__type_support.cpp:602:6: note: previous declaration of ‘bool unique_identifier_msgs::msg::typesupport_fastrtps_cpp::cdr_deserialize(eprosima::fastcdr::Cdr&, unique_identifier_msgs::msg::UUID&)’
 bool cdr_deserialize(
      ^~~~~~~~~~~~~~~
/home/yy/ws_test/build/example_interfaces/rosidl_typesupport_fastrtps_cpp/example_interfaces/action/dds_fastrtps/fibonacci__type_support.cpp:1163:8: warning: redundant redeclaration of ‘size_t unique_identifier_msgs::msg::typesupport_fastrtps_cpp::get_serialized_size(const UUID&, size_t)’ in same scope [-Wredundant-decls]
 size_t get_serialized_size(
        ^~~~~~~~~~~~~~~~~~~
[... 226 lines cut ...]
   ^
/home/yy/ws_test/build/example_interfaces/rosidl_typesupport_c/example_interfaces/action/fibonacci__type_support.cpp:183:51: note: previous declaration of ‘const rosidl_message_type_support_t* rosidl_typesupport_c__get_message_type_support_handle__example_interfaces__action__Fibonacci_GetResult_Request()’
 ROSIDL_TYPESUPPORT_INTERFACE__MESSAGE_SYMBOL_NAME(rosidl_typesupport_c, example_interfaces, action, Fibonacci_GetResult_Request)() {
                                                   ^
/home/yy/ws_test/build/example_interfaces/rosidl_generator_c/example_interfaces/action/fibonacci__type_support.h:125:3: warning: redundant redeclaration of ‘const rosidl_message_type_support_t* rosidl_typesupport_c__get_message_type_support_handle__example_interfaces__action__Fibonacci_GetResult_Response()’ in same scope [-Wredundant-decls]
   rosidl_typesupport_c,
   ^
/home/yy/ws_test/build/example_interfaces/rosidl_typesupport_c/example_interfaces/action/fibonacci__type_support.cpp:211:51: note: previous declaration of ‘const rosidl_message_type_support_t* rosidl_typesupport_c__get_message_type_support_handle__example_interfaces__action__Fibonacci_GetResult_Response()’
 ROSIDL_TYPESUPPORT_INTERFACE__MESSAGE_SYMBOL_NAME(rosidl_typesupport_c, example_interfaces, action, Fibonacci_GetResult_Response)() {
                                                   ^
/home/yy/ws_test/build/example_interfaces/rosidl_generator_c/example_interfaces/action/fibonacci__type_support.h:138:3: warning: redundant redeclaration of ‘const rosidl_service_type_support_t* rosidl_typesupport_c__get_service_type_support_handle__example_interfaces__action__Fibonacci_GetResult()’ in same scope [-Wredundant-decls]
   rosidl_typesupport_c,
   ^
/home/yy/ws_test/build/example_interfaces/rosidl_typesupport_c/example_interfaces/action/fibonacci__type_support.cpp:237:51: note: previous declaration of ‘const rosidl_service_type_support_t* rosidl_typesupport_c__get_service_type_support_handle__example_interfaces__action__Fibonacci_GetResult()’
 ROSIDL_TYPESUPPORT_INTERFACE__SERVICE_SYMBOL_NAME(rosidl_typesupport_c, example_interfaces, action, Fibonacci_GetResult)() {
                                                   ^
/home/yy/ws_test/build/example_interfaces/rosidl_generator_c/example_interfaces/action/fibonacci__type_support.h:151:3: warning: redundant redeclaration of ‘const rosidl_message_type_support_t* rosidl_typesupport_c__get_message_type_support_handle__example_interfaces__action__Fibonacci_FeedbackMessage()’ in same scope [-Wredundant-decls]
   rosidl_typesupport_c,
   ^
/home/yy/ws_test/build/example_interfaces/rosidl_typesupport_c/example_interfaces/action/fibonacci__type_support.cpp:265:51: note: previous declaration of ‘const rosidl_message_type_support_t* rosidl_typesupport_c__get_message_type_support_handle__example_interfaces__action__Fibonacci_FeedbackMessage()’
 ROSIDL_TYPESUPPORT_INTERFACE__MESSAGE_SYMBOL_NAME(rosidl_typesupport_c, example_interfaces, action, Fibonacci_FeedbackMessage)() {
                                                   ^
---

```

claireyywang, 2019-12-19:
Thanks for filing an issue! Confirming this is still a problem in Eloquent with all rmw implementations.

claireyywang, 2019-12-19:
Im also getting a lot of `[-Wclass-memaccess]` warnings which can be suppressed by adding `-Wno-class-memaccess` compile option in `CMakeList.txt`

claireyywang, 2019-12-19:
Thanks for filing an issue! Confirming that this is still a problem in Eloquent.