ros2/rosidl #422: [rosidl_generator_c] double free issue when initialization is failed

DongheeYe, 2019-11-20:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04 
- Installation type:
  - source 
- Version or commit hash:
  - 0.8.1 (36fac1e367bd98a493a7a0935b2c7b5ae86f5e7d)
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
``` sh
colcon build --symlink-install --merge-install 
```

#### Expected behavior
For example, build/std_msgs/rosidl_generator_c/std_msgs/msg/string__functions.c
```code 
// generated code
std_msgs__msg__String *
std_msgs__msg__String__create()                          // (1)
{
  std_msgs__msg__String * msg = (std_msgs__msg__String *)malloc(sizeof(std_msgs__msg__String));
  if (!msg) {
    return NULL;
  }
  memset(msg, 0, sizeof(std_msgs__msg__String));
  bool success = std_msgs__msg__String__init(msg);        // (2)
  if (!success) {                                         // (6) success is false
    free(msg);                                            // (7) double free msg
    return NULL;
  }
  return msg;
}

bool
std_msgs__msg__String__init(std_msgs__msg__String * msg)
{
  if (!msg) {
    return false;
  }
  // data
  if (!rosidl_generator_c__String__init(&msg->data)) {      // (3) return false
    std_msgs__msg__String__destroy(msg);                    // (4) call destroy
    return false;
  }
  return true;
}

void
std_msgs__msg__String__destroy(std_msgs__msg__String * msg)
{
  if (msg) {
    std_msgs__msg__String__fini(msg);
  }
  free(msg);                                                 // (5) free msg
}

```
There is no double free.

#### Actual behavior
There is no double free when initialization is failed.

#### Additional information
I thought if initialization is failed, then just should call fini.

dirk-thomas, 2019-11-20:
Can you please share some steps to reproduce the double free.

DongheeYe, 2019-11-21:
Please refer to the number comment of code.

(1) User calls std_msgs__msg__String__create. 
(2) std_msgs__msg__String__init will be called.
(3) rosidl_generator_c__String__init can return false. (Refer to the below code)
(4) Then, std_msgs__msg__String__destroy will be call.
(5) msg object will be freed in std_msgs__msg__String__destroy and return false.
(6) std_msgs__msg__String__init will return false, and success variable will be false in std_msgs__msg__String__create.
(7) msg object free will be call one more time. (double free)

```code
// rosidl/rosidl_generator_c/src/string_functions.c
bool
rosidl_generator_c__String__init(rosidl_generator_c__String * str)
{
  if (!str) {
    return false;
  }
  str->data = malloc(1);
  if (!str->data) {
    return false;
  }
  str->data[0] = '\0';
  str->size = 0;
  str->capacity = 1;
  return true;
}
```

closed 2019-12-03 by clalancette