ros2/rosidl #429: [rosidl_parser] Some IDL attributes cause rosidl_parser to fail

kylemarcey, 2020-01-03:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04
- Installation type:
  - Source
- Version or commit hash:
  - ROS2 Dashing
- DDS implementation:
  - RTI Connext
- Client library (if applicable):
  - 

#### Steps to reproduce issue

Use the following IDL and try to use it in a message package. It will fail before it gets to the DDS code generator, so you can reproduce this even if you do not have a DDS implementation that supports this particular annotation
```
module testmsgs {
  module msg {
    @transfer_mode(SHMEM_REF)
    struct IdlMessage {
      @key
      int32 int_value;
    };
  };
};
```

#### Expected behavior
Building a package that contains this message definition should succeed

#### Actual behavior
```
--- stderr: std_msgs                            
Tree(const_expr, [Tree(or_expr, [Tree(xor_expr, [Tree(and_expr, [Tree(shift_expr, [Tree(add_expr, [Tree(mult_expr, [Tree(unary_expr, [Tree(primary_expr, [Tree(scoped_name, [Token(ID
ENTIFIER, 'SHMEM_REF')])])])])])])])])])]) /home/kyle.marcey/ros2/src/ros2/common_interfaces/std_msgs/msg/IdlMessage.idl
Error processing idl file: /home/kyle.marcey/ros2/src/ros2/common_interfaces/std_msgs/msg/IdlMessage.idl
Traceback (most recent call last):
  File "/home/kyle.marcey/ros2/install/rosidl_generator_c/share/rosidl_generator_c/cmake/../../../lib/rosidl_generator_c/rosidl_generator_c", line 40, in <module>
    sys.exit(main())
  File "/home/kyle.marcey/ros2/install/rosidl_generator_c/share/rosidl_generator_c/cmake/../../../lib/rosidl_generator_c/rosidl_generator_c", line 35, in main
    args.generator_arguments_file,
  File "/home/kyle.marcey/ros2/install/rosidl_generator_c/lib/python3.6/site-packages/rosidl_generator_c/__init__.py", line 37, in generate_c
    generate_files(generator_arguments_file, mapping)
  File "/home/kyle.marcey/ros2/install/rosidl_cmake/lib/python3.6/site-packages/rosidl_cmake/__init__.py", line 90, in generate_files
    raise(e)
  File "/home/kyle.marcey/ros2/install/rosidl_cmake/lib/python3.6/site-packages/rosidl_cmake/__init__.py", line 70, in generate_files
    idl_file = parse_idl_file(locator)
  File "/home/kyle.marcey/ros2/install/rosidl_parser/lib/python3.6/site-packages/rosidl_parser/parser.py", line 64, in parse_idl_file
    content = parse_idl_string(string, png_file=png_file)
  File "/home/kyle.marcey/ros2/install/rosidl_parser/lib/python3.6/site-packages/rosidl_parser/parser.py", line 73, in parse_idl_string
    content = extract_content_from_ast(tree)
  File "/home/kyle.marcey/ros2/install/rosidl_parser/lib/python3.6/site-packages/rosidl_parser/parser.py", line 142, in extract_content_from_ast
    annotations = get_annotations(struct_defs[0])
  File "/home/kyle.marcey/ros2/install/rosidl_parser/lib/python3.6/site-packages/rosidl_parser/parser.py", line 517, in get_annotations
    value = get_const_expr_value(const_expr)
  File "/home/kyle.marcey/ros2/install/rosidl_parser/lib/python3.6/site-packages/rosidl_parser/parser.py", line 527, in get_const_expr_value
    assert len(literals) == 1, str(const_expr)
AssertionError: Tree(const_expr, [Tree(or_expr, [Tree(xor_expr, [Tree(and_expr, [Tree(shift_expr, [Tree(add_expr, [Tree(mult_expr, [Tree(unary_expr, [Tree(primary_expr, [Tree(scoped
_name, [Token(IDENTIFIER, 'SHMEM_REF')])])])])])])])])])])
make[2]: *** [rosidl_generator_c/std_msgs/msg/idl_message.h] Error 1
```

#### Additional information

I believe the problem is that the function `get_const_expr_value` in `rosidl_parser/parser.py` only handles `literals`, but the grammar allows either a `literal` or a `scoped_name`

kylemarcey, 2020-01-03:
I hacked around a bit on this and was able to find a potential fix, but I am not familiar enough with this code to know if this change will affect anything else. This is a patch with those changes:
[parser.patch.txt](https://github.com/ros2/rosidl/files/4021207/parser.patch.txt)

dirk-thomas, 2020-01-04:
> This is a patch with those changes:

Please don't use patch files but create a pull request with the proposed changes.

kylemarcey, 2020-01-07:
@dirk-thomas I have converted the patch to a PR.

PR #430 (2020-01-16): Allow 'get_const_expr_value' to parse either literals or scoped_names…
- Changed the function `get_const_expr_value` in `parser.py` to work when the const expr is a scoped_name as well as when it is a literal

Fixes #429.
closed 2020-01-16 by dirk-thomas