ros2/rosidl #437: Should rosidl_generator_c be a rosidl_runtime_packages?

ahcorde, 2020-02-18:
`rosidl_generator_c` is not a member of **rosidl_runtime_packages** but `rosidl_generator_cpp` is a member of **rosidl_runtime_packages**

Not sure about why this is like this, any thoughts @wjwwood ?

wjwwood, 2020-02-20:
I don't, it seemed fishy to me though. Maybe you can do a bit more research into when `rosidl_generator_cpp` was added and why the c version was not.

@dirk-thomas may have some insight as well.

dirk-thomas, 2020-02-20:
It probably should be listed as a explicit dependency in `rosidl_runtime_packages` for consistency. Since the C++ as well as the Python generators both already export a dependency on `rosidl_generator_c` it won't make a difference though.

ahcorde, 2020-02-27:
closing this, Fixed in this PR https://github.com/ros2/rosidl/pull/440/files

closed 2020-02-27 by ahcorde