ros2/rosidl #443: Splitted rosidl_generator_c and rosidl_generator_cpp in two: rosidl_generator_x and rosidl_runtime_x

ahcorde, 2020-03-04:
This issue is to track all the PRs that are need it to compile ros2.repos with the changes introduced in this PR split https://github.com/ros2/rosidl/pull/442. We will have these packages:

 - rosidl_generator_c
 - rosidl_generator_cpp 
 - rosidl_runtime_c (new)
 - rosidl_runtime_cpp (new)

ahcorde, 2020-03-04:
- [ ] https://github.com/ros2/rmw_cyclonedds/pull/110
 - [ ] https://github.com/ros2/rmw_opensplice/pull/303
 - [ ] https://github.com/ros2/rmw_fastrtps/pull/351
 - [ ] https://github.com/ros2/rmw_implementation/pull/86
 - [ ] https://github.com/ros2/rosidl_python/pull/100
 - [ ] https://github.com/ros2/rosidl_typesupport_fastrtps/pull/32
 - [ ] https://github.com/ros2/rosidl_typesupport/pull/58
 - [ ] https://github.com/ros2/rosbag2/pull/310
 - [ ] https://github.com/ros2/rosidl_defaults/pull/4

dirk-thomas, 2020-03-04:
It would make it easier to review these PRs if you could reference at least one specific source line in the packages where new dependencies are added to clarify why they are added. It might also be good to clarify - when runtime packages are added - why generators are still needed.

ahcorde, 2020-03-04:
> It might also be good to clarify - when runtime packages are added - why generators are still needed.

This was my fault I didn't remove the rosidl_generator_c/cpp dependencies. I revisited all the changes and removed this dependencies

ahcorde, 2020-03-05:
I have been checking rcl, rcl_lifecycle, rcl_action, rclcpp and rclcpp_action all of them make use of rosidl_generator ( with the new packages should be rosidl_runtime) but I think this dependecy is not need it, the reason why it's because rosidl_generator/rosidl_runtim is used where a message packages it's being used too.

Some examples:
RCL:
https://github.com/ros2/rcl/blob/master/rcl_action/CMakeLists.txt#L57
https://github.com/ros2/rcl/blob/master/rcl_lifecycle/CMakeLists.txt#L48

RCLCPP:

https://github.com/ros2/rclcpp/blob/master/rclcpp_action/CMakeLists.txt#L36
https://github.com/ros2/rclcpp/blob/master/rclcpp/CMakeLists.txt#L180

@wjwwood and @dirk-thomas what do you think?

ahcorde, 2020-03-23:
testing all PRs

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

ahcorde, 2020-03-24:
- [ ] https://github.com/ros2/rmw_connext/pull/399
 - [ ] https://github.com/ros2/rosidl_typesupport_connext/pull/49

ahcorde, 2020-03-24:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

dirk-thomas, 2020-03-24:
Please double check that with all the PRs the actual goal (avoiding a runtime dependency on any generator package) has been achieved.

ahcorde, 2020-03-26:
`rmw_connext` and `rosidl_typesupport_connext` approved. Launching CI for all the commits:

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

ahcorde, 2020-03-26:
I have been reproducing the failure in my local MAC OS machine. if I used `include_directories` there is no issue when cmake is trying to find the generated headers, but with `target_include_directories`, I think a race condition is happening. If I execute colcon a second time then the error disappears.

@wjwwood or @dirk any thoughts about this?

dirk-thomas, 2020-03-26:
The `rosidl_generator_cpp` PR removes the `add_dependencies(<test_target> ${PROJECT_NAME})` calls for all test targets: e.g. https://github.com/ros2/rosidl/pull/442/files#diff-59c331d207ef4d5e6a390e74e0bc4739L54

Without that dependency CMake tries to build the test before the code generator was run which creates the used header files.

ahcorde, 2020-03-27:
The packages depend on `rosidl_generator_c` or `rosidl_generator_cpp` are:
 - rosidl_typesupport_introspection_cpp
 - rosidl_typesupport_introspection_c
 - rosidl_default_generators

The packages depend on `rosidl_runtime_c` or `rosidl_runtime_cpp`

 - rmw_cyclonedds
 - rmw_fastrtps
 - rmw
 - rosidl_python
 - rosidl_typesupport_fastrtps
 - rosidl_typesupport
 - rosbag2
 - rosidl_defaults
 - rclcpp
 - rcl

dirk-thomas, 2020-03-27:
> The packages depend on rosidl_generator_c or rosidl_generator_cpp are:

I guess the dependencies still need to be removed from the two introspection packages then to achieve the stated goal, right?

ahcorde, 2020-03-30:
- Removed `rosidl_generator_c`  and `rosidl_generator_cpp` from `rosidl_typesupport_introspection_cpp` and `rosidl_typesupport_introspection_c`.
 - Added `rosidl_generator_c`  and `rosidl_generator_cpp` to `rosidl_generator_py`

------

The packages depend on `rosidl_generator_c` or `rosidl_generator_cpp` are:
 - rosidl_default_generators
 - rosidl_generator_py

The packages depend on `rosidl_runtime_c` or `rosidl_runtime_cpp`

 - rmw_cyclonedds
 - rmw_fastrtps
 - rmw
 - rosidl_python
 - rosidl_typesupport_fastrtps
 - rosidl_typesupport
 - rosidl_typesupport_introspection_cpp
 - rosidl_typesupport_introspection_c
 - rosbag2
 - rosidl_defaults
 - rclcpp
 - rcl

ahcorde, 2020-04-06:
- [ ] https://github.com/ros2/rmw_dds_common/pull/10

ahcorde, 2020-04-06:
The packages depend on `rosidl_generator_c` or `rosidl_generator_cpp` are:

- rmw_dds_common_generators
- rosidl_default_generators
- rosidl_generator_py

The packages depend on `rosidl_runtime_c` or `rosidl_runtime_cpp`

- rcl
- rclcpp
- rmw
- rmw_cyclonedds
- rmw_dds_common
- rmw_fastrtps
- rosbag2
- rosidl_defaults
- rosidl_python
- rosidl_typesupport
- rosidl_typesupport_fastrtps
- rosidl_typesupport_introspection_c
- rosidl_typesupport_introspection_cpp

dirk-thomas, 2020-04-06:
Please update all PRs to update / revert changes to dependency types of the follow kind, e.g. ros2/rmw_dds_common#10:

Since the added `buildtool_depend` and `buildtool_export_depend` dependencies are a build as well as a run dependency please revert the dependency type to the previously used `depend`. I would rather like to avoid that kind of dependency type changes in this already massive amount of changes.

dirk-thomas, 2020-04-06:
Also please update the list of all related PRs somewhere in this ticket.

ahcorde, 2020-04-07:
The packages depend on `rosidl_generator_c` or `rosidl_generator_cpp` are:

- rmw_dds_common_generators https://github.com/ros2/rmw_dds_common/pull/10
- rosidl_default_generators https://github.com/ros2/rosidl_defaults/pull/5
- rosidl_generator_py https://github.com/ros2/rosidl_python/pull/100

The packages depend on `rosidl_runtime_c` or `rosidl_runtime_cpp`

- rcl https://github.com/ros2/rcl/pull/588
- rclcpp https://github.com/ros2/rclcpp/pull/1014
- rmw  https://github.com/ros2/rmw/pull/198
- rmw_cyclonedds https://github.com/ros2/rmw_cyclonedds/pull/138
- rmw_dds_common https://github.com/ros2/rmw_dds_common/pull/10
- rmw_conenxt https://github.com/ros2/rmw_connext/pull/399
- rmw_fastrtps https://github.com/ros2/rmw_fastrtps/pull/351
- rosbag2 https://github.com/ros2/rosbag2/pull/310
- rosidl  https://github.com/ros2/rosidl/pull/442
  - rosidl_runtime_c
  - rosidl_runtime_cpp
  - rosidl_typesupport_introspection_c
  - rosidl_typesupport_introspection_cpp
- rosidl_defaults https://github.com/ros2/rosidl_defaults/pull/5
- rosidl_python https://github.com/ros2/rosidl_python/pull/100
- rosidl_typesupport https://github.com/ros2/rosidl_typesupport/pull/58
- rosidl_typesupport_connext https://github.com/ros2/rosidl_typesupport_connext/pull/49
- rosidl_typesupport_fastrtps https://github.com/ros2/rosidl_typesupport_fastrtps/pull/32

-------
## update / revert changes to dependency types

 - rmw_connext https://github.com/ros2/rmw_connext/pull/399/files
 - rosidl_typesupport_connext https://github.com/ros2/rosidl_typesupport_connext/pull/49/files
 - rosbag2 https://github.com/ros2/rosbag2/pull/310/files
 - rmw_fastrtps https://github.com/ros2/rmw_fastrtps/pull/351/files
 - rmw_dds_common https://github.com/ros2/rmw_dds_common/pull/10/files
 - rcl https://github.com/ros2/rcl/pull/588
 - rosidl_typesupport_fastrtps https://github.com/ros2/rosidl_typesupport_fastrtps/pull/32/files
 - rosidl https://github.com/ros2/rosidl/pull/442/files

ahcorde, 2020-04-08:
These warnings on Windows were already fixed, but I didn't update the branch in the PR

```
15:49:21 C:\ci\ws\src\ros2\rmw_dds_common\rmw_dds_common\test\test_graph_cache.cpp(195,8): warning C4996: 'strncpy': This function or variable may be unsafe. Consider using strncpy_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details. [C:\ci\ws\build\rmw_dds_common\test_graph_cache.vcxproj]

15:49:21 C:\ci\ws\src\ros2\rmw_dds_common\rmw_dds_common\test\test_graph_cache.cpp(388,8): warning C4996: 'strncpy': This function or variable may be unsafe. Consider using strncpy_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details. [C:\ci\ws\build\rmw_dds_common\test_graph_cache.vcxproj]

```

ahcorde, 2020-04-08:
Testing changes on rmw_cyclonedss

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

ahcorde, 2020-04-09:
- Updated all repos involved in this change. 
 - Skipping `ros1_bridge`, `qt_*`, `rqt_*`, and `rviz_*` packages. 
 - Testing against `test_rclcpp` and `rosbag2_tests`.

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

ahcorde, 2020-04-09:
- One warning made the builds unstable but there is a ([PR pending](https://gitlab.com/micro-ROS/ros_tracing/ros2_tracing/-/merge_requests/163))

In MacOS
 - projectroot.cppcheck

Windows
 - some failures in test_rclcpp

```
CMake Deprecation Warning at /home/jenkins-agent/workspace/ci_linux-aarch64/ws/install/ament_cmake_export_interfaces/share/ament_cmake_export_interfaces/cmake/ament_export_interfaces.cmake:37 (message):
ament_export_interfaces() is deprecated, use ament_export_targets() instead
Call Stack (most recent call first):
CMakeLists.txt:85 (ament_export_interfaces)
```

closed 2020-04-10 by dirk-thomas